c0pperdragon/C64-Video-Enhancement #1: Repository appears to be incomplete

ikorb, 2019-03-10:
Hi,

it appears that the repository is incomplete - I can only see KiCAD files for the FPGA board, but not for the VIC-II adapter board. Furthermore, anything related to the FPGA's contents seems to be missing completely.

There also seems to be no licensing information in the repository, but I guess this is related to the fact that the other data is also missing.

c0pperdragon, 2019-03-10:
All this is located in my repository: A-Videog Board in the subfolder C64
mod. This board here is softwarecompatible with the original design and
uses the same adapterboard.  I will add some info on the page and also a
proper licence for free use.

ikorb, 2019-03-10:
Found it, thanks!

closed 2019-03-10 by ikorb