c0pperdragon/C64-Video-Enhancement #2: Production progress and delivery

c0pperdragon, 2019-03-15:
To keep everybody informed about the situation of my production efforts, I will regularly update this thread. 

As you probably guessed, I did not expect such a demand and I did not even have the neccessary parts ready. Of course I now ordered enough to produce 10 kits, and this delivery should arrive in the next days. In the meantime I did scrape together my last pieces and could create just 1 VIC adapter board (which is basically half a kit). 

A collegue agreed to let me use his reflow oven and solder paste to speed up a small part of the assembly, namely the underside of the FPGA board. For the rest I still need to manually solder everything. Until we can do that (the solder paste stencil may be a bit tricky to make), I will concentrate on making the adapter boards. So I will not get a complete kit together before at least two weeks or so. 

Anyway whenever I have complete kits, I will send them to you in order of payment.
I will send out individual mails notifying you about your position on the list.

Adapter boards: 1  FPGA boards: 0

c0pperdragon, 2019-03-20:
Disaster: My main C64 (a 1986 breadbox) just went to a black screen. This leaves me with only a C64C which I had turned into a NTSC machine for use of making the firmware NTSC compatible. 
Anyway this does not hinder me to create and test the actual hardware for the mod, but I can not improve on the firmware features for PAL right now.  With some luck it is a simple fix (the PLA being the first suspect), but I will surely need replacement parts.
Anyway, building of the hardware progresses as planned.

Adapter boards: 3  FPGA boards: 0

bodgit, 2019-03-20:
I found this project via [oshpark.com](https://oshpark.com/). I'm interested in one of these, are orders being taken somewhere? Or have I missed the boat?

Also, would this be feasible to adapt to a C128 (for the 40-column output)?

c0pperdragon, 2019-03-21:
Please leave this issue thread for announcements concerning the current production efforts. You can open any number of new issues for questions.

By the way: Production is progressing just find.
Adapter boards: 5 FPGA boards: 0

c0pperdragon, 2019-03-23:
Busy weekend.
Adapter boards: 10 FPGA boards: 0

c0pperdragon, 2019-03-31:
Building the FPGA boards is a bit more time-consuming than the adapter boards. I finished and tested two this weekend.  But I am not satisfied with the firmware yet, as it still has incompatibilities in the border engine, so I can not ship the kits yet. To get this right, I need the replacement part for my C64(PAL).
Adapter boards: 10  FPGA boards: 2  Shipped: 0

that1crzywhtguy, 2019-04-05:
Will there be another preorder? If so where?

c0pperdragon, 2019-04-06:
Bad news is, the replacement part for my C64 (PAL) does not solve the black screen issue. Good news is, I can just put a PAL VIC into my NTSC machine to make it kinda work like PAL, but with a slightly higher frame rate and with a totally screwed choma output. This now allowed me to get the firmware part for the border logic finally right.
Build progresses just fine, and with two more FPGA boards and the newest firmware, I have new 4 complete kits ready for shipping on monday.
The first 4 recipients on my list will receive their tracking code right afterwards.

Adapter boards: 10 FPGA boards: 4 Shipped: 0

[image]

MagerValp, 2019-04-07:
Iirc, just swap the crystal as well and bridge the jumper and you'll have a complete PAL conversion.

Great to see the progress!

c0pperdragon, 2019-04-15:
After a busy week, I have now shipped 4 more kits. I will send out individual mails with postage tracking codes today at the evening.

This now completes the backlog of all 8 sold kits.
As I have now parts left to build 2 more kits, I can probably sell these also (given, that I don't screw up on this last mile).  As demand is possibly higher than the supply, everyone who wants to get one of these kits can apply for one right in this thread here. Instead of selling to the highest bidder, I will do a small lottery to decide who gets the option to buy one at the previous terms: € 100 including delivery to the EU. €105 for delivery outside.
Application ends on 20.4.2019, 24:00 MEST.

Additionally I have 2 sets of spare PCBs (one for the VIC adapter, one for the FPGA board) left over. If anyone wants to have any of these (or both sets), I would give them away for parts cost + postage (€13 in EU, €18 for outside). No lottery here, these go to the first persons responding.

Adapter boards: 10 FPGA boards: 8 Shipped: 8

closed 2019-04-22 by c0pperdragon