c0pperdragon/C64-Video-Enhancement #4: C128 support or version

bodgit, 2019-03-21:
I presume it should be possible to do the same video enhancement on a C128? As I understand it, the VIC-IIe chip has 8 additional pins for a total of 48, however they're apparently for the Z80 CPU and the extra keys on a C128.

I'm guessing the modulator is going to be a different part so there might also be some changes needed for everything to line up such as the TRRS jack and mode switch to line up with the holes in the case. I can check this on my C128 unless you have one to hand.

c0pperdragon, 2019-03-21:
At a quick glance on the pinout diagram of the VIC-IIe, there seem to be some substantial differences to the VIC-II. As my mod totaly relies on the precise signal timings of the VIC, this will surely not work with the current firmware.

The modulator replacement board needs also a complete rework, because it will not fit in the space where the modulator is. The two video chips are just too close.

As the VIC adapter board will naturally also not fit, there is basically nothing (not hardware, not firmware) that could actually be re-used from my mod.

Sorry

bodgit, 2019-03-25:
Is the VIC-IIe that different? There are extra pins, which according to the Wikipedia article are there for supporting the Z80 and the extra keys on a C128, but put there because it was the easiest place to put them, absolutely nothing to do with video output. Surely C64 compatibility wouldn't work if it behaved differently to a "normal" VIC-II?

I understand that the VIC adapter would need reworking to match the different pin count and layout, but I'd be happy to try and do that and attach it to a loose C64 modulator board and see if it at least works. Would that not work?

c0pperdragon, 2019-03-25:
Well, I can't even guess if this could somehow work. My firmware heavily depends on the signals to arrive exactly as the VIC-II and the CPU generates them.
Also having the signals DMAA/ and DMAR/ on the VIC-IIe hints at some different kind of Voodoo than was actually used in the C64 for sharing the RAM with the CPU.
Anyway, you are free to try anything you like, but I won't go down that rabbit hole.

bodgit, 2019-03-26:
Understood. First things first, I should get something working for my C64.

c0pperdragon, 2020-01-28:
Due to additional requests I am now in the phase of considering to do something for Commodore 128 users.  If you are interested, let me know here so I can come up with some plan of attack on the problem.

jpino1, 2020-01-28:
Definitely very interested and grateful if done finally I have three C128

c0pperdragon, 2020-01-28:
The main problem with this is of course that I do not have a C128 myself and I also do not want to buy one. So everything I can say  is purely theoretical.

Looking at the pinout of the VIC-IIe, all the signals from the VIC-II are also present and maybe they also work in the same way. If this is the case, then the mod could actually work for the 40-column mode (the one with the VIC-IIe generating the picture).
To try this out, I could create an adapter board to plug between VIC-IIe and the mainboard to tap into all the signals exactly as for the C64. This could then connect to the FPGA board which could be directly used without any changes. A first attempt would just require temporarily connecting everything together without actually removing the RF modulator.

Permanently installing the FPGA board into the C128 is surely more complicated because there is not enough space. Some part of the shielding from the video chip enclosure needs to be cut away to make the FPGA board fit into the place of the RF modulator.  

For this whole attempt, I would need a volunteer who is desperate enough to try it ;-)
What you would need to do: As soon as it is readily available, get a C64 video enhancement board. There is a fair chance you can make an agreement with VGP so they take it back if this whole thing does not work out.
I will design a C128 adapter board that _should_ work regarding to specs. As I have no way to really test it, I can make absolutely no promises. For the cost of the parts and shipping, I can send it for you to assemble and try. If it works, its fine. If it doesn't work, we could try to do some error tracking, but on a distance this is really difficult. In the worst case you lose the money for the adapter. In the very worst case, you break your machine in the attempt.

So, anyone there to take the risk?

ChrisToast89, 2020-01-29:
I have a standard C128 (currently with a LumFix128). I have board assembly skills (no BGA), I would be very interested in a component fix as it is my "daily driver" 64 essentially. However, I have zero VHDL skills. I would be happy to help assemble and test.

jpino1, 2020-01-29:
For my part there would be no problem I have a C128 PCB ready for testing.
All my C128 have been recovered and repaired by me.
[image]

I could mount the FPGA board I have some PCBs but I need someone to design the adapter and I am not sure I can do one board with my CNC could try is the fast way. If not, you should ask a PCB manufacturer and make a few units to see if they work. Once we have everything I can mount it on my C128 pcb test. I'm going to order the components.
I was also looking at the VIC-II signals in the design of the C64 and I think it should work in the C128, making the adapter. I put a comment it here on GitHub somewhere some time ago, in fact it is something I wanted try to do myself.

jpino1, 2020-01-29:
Here is my old comment about it:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/7#issuecomment-549088029

Let's do it the C128 deserves it :-)

c0pperdragon, 2020-01-29:
Very good then. So this is the plan now: I will design the adapter board (basically just re-routing the traces to the correct VIC-IIe pins) and publish the designs. For anyone who had already successfully built the C64 boards, using this should then be quite straight-forward.

If the VIC-IIe and the relevant parts of the system really do behave like the VIC-II, the latest firmware should actually work. 
There is one issue though: Detection of NTSC is done by measuring the CPU clock speed during the first few millisecond after power-on. If the video chip is not generating the 1.02 MHz frequency right then, the mod will remain in PAL mode. I guess the default behavior of the chip may indeed be to generate the clock signal, so everything could be OK in this respect. 
Still I would prefer to have initial tests done with a PAL machine before tackling this specific issue.

c0pperdragon, 2020-01-29:
Hmm, maybe I got the details of NTSC detection wrong. The mod actually re-checks for NTSC every few milliseconds or so. So chances are very high that it will work for NTSC just as well as for PAL.

c0pperdragon, 2020-01-29:
@jpino1 :  As a kind of research, could you use your board to check if the pins of the RF modulator are at the exact position that the FPGA board expects? I guess this should be the case, as it looks exactly as the RF modulator built into he short C64 boards.
This can easily be done from the underside of the main board,  where you need to check if the 
connectors RFCONC1, RFCONC2, GND5 on the FPGA board line up with the pins and the mounting cutout for the metal shield.

Also it would be interesting to know how far the FPGA board would extend into the metal shielding of the graphics chips.

jpino1, 2020-01-29:
He took some pictures.
RFCON1 RFCON2 match, no problem here.
There are some unwanted contact points. I think we could solve it by mounting the PCB a little higher to save the shield and mount the jack from the bottom I think it could match. maybe you should file a bit of the DIN connector...
[image]
[image]
[image]

c0pperdragon, 2020-01-29:
Making the board sit so high will very probably cause trouble with the ribbon cable connector not fitting into the case. Also mounting the jack from the other side will switch the Pb and Pr signals (which may not be such a big deal).
Another issue is of course the exact position of the hole in the case where the TV cable was connected. It should be properly aligned in height with the video jack. And the mode switch should also be accessible through the rectangular hole.

Anyway, I guess the board will fit after removing the obstacles (cut down the shielding, cut a slit into the DIN connector). To avoid shorting to the metal shielding, just put some insulation tape on the underside of the board there. As there are no components in this place, this is easy.
Maybe tilting the board a little to better reach the holes could help.

c0pperdragon, 2020-01-31:
Today I designed the adapter board.  It was quite tricky, because in the C128, there is this big capacitor sitting right where the connector should go. So I moved the connector to one side to avoid clashing. Note, that there are rare variants of the C128 board (the C128DRC) where this capacitor is again the way, so you would need to relocate it in this case.

Everything (gerber files, BOM) is available at
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/c128board

Good luck, and keep me informed!

pazuzu72, 2020-07-27:
Given the horrible video quality of the C128 (I love it on C64 too) it would be great to use it on this computer. If anyone did it,  it would be nice to know if it works or not.  I would do it in an external version since I already have a module mounted and working (A-Video board).  So I only need the PCB to put under the VIC-II of the C128.

djcarlin, 2020-09-25:
Anyone every try this?  I'm real happy with component mod on C64, would love something similar to clean up C128 video.

JDet-74, 2020-12-30:
Hey c0pperdragon, 
what about the C128 project?
did you continue working on it?
I myself have a small project to connect both video outputs of the C128 to a display device.
It was created because of my 30 year old SCART cable on which I can switch between 40 characters and 80 characters mode on the SCART connector. My repository is called "128toSCART_SMT".
Unfortunately, I have to say that the SCART interface on modern display devices is not the best choice. My project converts the RGBI of the 80 character mode to RGB. You can then choose between 40 col. Mode (CVBS or S-Video) and 80 col. Switch mode (RGB). Your variant is of course a better solution.
I guess that it is definitely possible to process the VIC-IIe and the VDC with an FPGA and to realize a component output.
If you continue to pursue your project and it helps you, then I would provide you with a C128 DCR mainboard.
I also have a C=KEY board lying around that I have never put into operation. You could use this to connect a conventional keyboard. If you want to give me your address please contact me. Then I will send you the C128 DCR (PAL) board and the C = KEY board. I am very interested in such a solution myself. Unfortunately I have no experience with FPGA and VCL (not yet;)

c0pperdragon, 2020-12-31:
Actually there is some movement in the C128 issue right now. I am just waiting for a delivery of PCBs to make the adapter. One C128 user volunteered to pay the expenses and try the adapter in his machine.  Maybe this will work already.

ami-commo-fan, 2021-01-10:
I have now it working in my Commodore 128 thanks to c0pperdragon willing to build me the C128 adapter (Thank you again!).
Thankfully I connected it right finaly and it worked perfectly. Nice clean picture just like on C64. Of course the C64 enhancement board will not fit inside because the 128 modulator is smaller, but that does not matter for me. And here some pictures.
[url=https://i.ibb.co/YN9zdV1/IMG-8550-25.jpg][img]https://i.ibb.co/yNTwkxZ/IMG-8550-25.jpg[/img][/url]
[url=https://i.ibb.co/LnZy6xW/IMG-8551-25.jpg][img]https://i.ibb.co/721DVn5/IMG-8551-25.jpg[/img][/url]
[url=https://i.ibb.co/wyWTR6G/IMG-8554-25.jpg][img]https://i.ibb.co/PxhRNCX/IMG-8554-25.jpg[/img][/url]

pazuzu72, 2021-01-11:
Super mega great news!!!!! 
Thank you for sharing. C128 video quality  is so terrible, ne beneficia molto più del C64.

c0pperdragon, 2021-01-11:
Yes, I actually have 15 PCBs left.  I only needed one, but my PCB manufacturer charges me always at least 10 pieces and then also adds some extra.

Actually these PCBs were paid for by @ami-commo-fan  . But as I basically built his adapter for free, I guess it is OK that I sell them off now.  

So you can have 2 PCBs (just in case you screw one up) for €5 including international delivery in a simple non-tracked envelope. Paypal account  reinhard.grafl@aon.at.   Please add the shipping address.

pazuzu72, 2021-01-11:
Is the pcb 0.8mm?

c0pperdragon, 2021-01-12:
It is 1.0 mm. I used this instead of 0.8 to make it a bit more stable.  This seems to be perfect in conjunction with the long-tailed IC sockets.

pazuzu72, 2021-01-12:
> It is 1.0 mm. I used this instead of 0.8 to make it a bit more stable. This seems to be perfect in conjunction with the long-tailed IC sockets.

Payment sent. Do you have a link for the socket?

c0pperdragon, 2021-01-12:
The BOM is included in the documentation with part numbers and all:
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/c128board

The IC sockets can be found here, fore example:
https://www.mouser.at/Search/Refine?Keyword=111-93-640-41-001000
https://www.mouser.at/Search/Refine?Keyword=111-93-308-41-001000

c0pperdragon, 2021-01-12:
PCBs sent. Good luck with your build.

fortiko, 2021-01-12:
Hello @c0pperdragon I understand that this should also work for a C128D, correct? And would you also sell completely assembled adapters, for what price? Thank you in advance!

c0pperdragon, 2021-01-12:
Yes, as far as I know, the board layout should be quite the same.
There is a different variant (C128D-CR) which has a different board layout and there is a big capacitor which may collide with the adapter.

I don't intend to sell assembled adaptors in the near future. You can have bare PCBs, though. I still have 13.

pazuzu72, 2021-01-12:
> PCBs sent. Good luck with your build.

Thanks!!

JDet-74, 2021-01-13:
hello c0pperdragon,
do you only have the adapter boards of the c128 variant? Did you design two variants? One for the C128 and one for the 128D? I would also be interested in an adapter for my C128DCR. Do I understand correctly that I need the board under "C64-Video-Enhancement /" and the adapter under "A-VideoBoard / c64mod / c128board /"?
Where can I find the GERBER files for the board with the FPGA on it or do you still have circuit boards that you would sell me?
It would be great if you could tell me where the GERBER files for the FPGA board can be found (there are gerber files in the c128board directory). Otherwise
I would buy boards from you and try a solution on my C128DCR. Thanks for your help and you are doing a great job.

c0pperdragon, 2021-01-13:
I am not doing any assembled boards for this project right now. I do have spare PCBs for the C128 adapter, but nothing else.
Everything to build the FPGA board is at:  https://github.com/c0pperdragon/C64-Video-Enhancement/tree/master/doc
Everything for the C128 adapter is at: https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/c128board
Note that you can still get assembled FPGA boards (as part of the C64 component video mode) from videogameperfection.com

JDet-74, 2021-01-13:
Alright. 
I will use the Gerber files and order some prototype PCBs in the  next couple days.
It could be that the FPGA board fits as a modulator replacement cause there is more space in the C128DCR. I will give it a try building the stuff together by myself. I think you mentioned a USB-Blaster for programming the FPGA in another thread.
I will post my success or failure if the World is done. ✌😉

c0pperdragon, 2021-01-13:
Guessing from pictures of this board, there may be just enough space to make the FPGA board fit. 
On the other hand, you surely have to relacate C63 to make space for the adapter board (should not be too difficult).

If you use the FPGA board from videogameperfection.com, you probably don't need an USB-Blaster, as they come pre-programmed with firmware 2.6 which is probably working just fine.

Good luck!

JDet-74, 2021-01-13:
I am a crazy old man ... I will order the minimum quantity (5 pcs) of each PCB (FPGA, C128, C64) prototype PCBs from JLCPCB.
Furthermore, I already checked your BOMs and will order the Parts from MOUSER. Only the TRRS connector is a little tricky. You mentioned CONRAD electronic I think. When I have all parts together I will do one of each PCBs :) Doing some SMT work calms me down ;) Programming the FPGA with the USB-Blaster will be a challenge, but that doesn't matter.
I will post some pictures and status messages if the time is come.

pazuzu72, 2021-01-15:
> PCBs sent. Good luck with your build.

Arrived!! Thank you

JDet-74, 2021-01-15:
Today I printed out the circuit boards and stuck them on cardboard. Then I placed them on the board. That doesn't look that good.

[image]
[image]
[image]
[image]

There is not enough space for the FPGA board. The FPGA board would have to be adjusted a little. Too bad. :(

The VIC adapter board then looks like this.

[image]
[image]
[image]
[image]

The electrolytic capacitor is in the way, as c0pperdragen said. If the plug-in adapter were 7mm further up it could fit.
A bigger problem is the power supply in the C128DCR. That's right above the VDC and VIC.
I have to think about it ... it should be feasible.

c0pperdragon, 2021-01-15:
If you don't want to change the PCB design for the FPGA board, you can as well just saw off the part that overlaps the RF shielding. On this specifc location are basically just the doubled connectors for the long board installation which are not needed here. There is also the via to GND for the coupling capacitor of the 3.3V rail. You could even saw away half of the via and it would still work.

c0pperdragon, 2021-01-15:
So save space above the VIC adapter board, you could use angled pin headers for the ribbon cable connectors.

JDet-74, 2021-01-15:
> If you don't want to change the PCB design for the FPGA board, you can as well just saw off the part that overlaps the RF shielding. On this specifc location are basically just the doubled [...]
I think I will change the PCB design, if I do not HURT your feelings.
RECON1 & RECON2 is not needed, as you said. What is about JFPC1? Are this pads needed?
I will try to shrink the FPGA Board to fit in the 128DCR ... it is worth a try ... why not.

> So save space above the VIC adapter board, you could use angled pin headers for the ribbon cable connectors.
You mention this HARTING (#9181209622) ribbon connector in the BOM. I think that is a good choice, cause it saves hight.
I will check the space between the PSU and the VIC. I will put the VIC in a 48 pin socket and then into the U21 socket.

[image]

If I am lucky the VIC will not touch the PSU ... If that works I change also the VIC adapter PCB to avoid the capacitor.
There is a lot to do ... ;) ... let's get started.

There is just one prob ... I am a EAGLE guy ... (hahaha) I work with EAGLE, with EasyEDA and now ... I have to learn KiCad ... funny.
It will take a while to change the design, but ... I don't care.

c0pperdragon, 2021-01-16:
That is a very good idea to modify the PCB for a better fit. I nearly would have offered to do it myself, but in this way it is much better. :-)
If you have experience with other PCB layouting programs, KiCad should not be so difficult to learn, I guess.

If you are now building the FPGA board explicitly for the C128 you can actually leave out some parts (as you guessed):
- RFCON1, RFCON2
- either of  GPIO1 or JFPC1  (you can opt for ribbon cable or flat flex cable to transport the signal from the adapter to the FPGA board)
- R36 and JPLUM1

When you redesign the adapter board already, you could also change cabling to flat flex to save space. I did never use this, but it seems to be quite popular.

If you want to create a unified solution that would be really useful for other C128 users as well, maybe you can find a way to redesign both boards in such a way to make it fit into other C128 models as well.  As I have no C128 model at all, I can not give you much hints there. I don't even know if it is possible at all, but I guess some users would be quite grateful for that.
Some possible ideas to shrink down the board: Use more of the buttom side. Indeed use a flat flex cable. You don't need the big heatsink for REG1 when you power the board with 5 volt.

Whatever you decide,
good luck!

JDet-74, 2021-01-16:
> That is a very good idea to modify the PCB for a better fit. I nearly would have offered to do it myself, but in this way it is much better. :-)

Yes ... that is a better way for you ;-)

> If you have experience with other PCB layouting programs, KiCad should not be so difficult to learn, I guess.

Yesterday I played around with KiCad - it's quite similar but also different :-)
I think I can manage it - but it will take longer than with eagle or EasyEDA. But I will do with KiCad - after finishing the work I can send it to you and you can include it in your repository. Or I can create another fork of your repository. Or I create a new repository and link to your FPGA source code. Whatever ...

> If you are now building the FPGA board explicitly for the C128 you can actually leave out some parts (as you guessed):
RFCON1, RFCON2
either of GPIO1 or JFPC1 (you can opt for ribbon cable or flat flex cable to transport the signal from the adapter to the FPGA board)
R36 and JPLUM1

I will have this in mind ... this will save space on the PCB.

> If you want to create a unified solution that would be really useful for other C128 users as well, maybe you can find a way to redesign both boards in such a way to make it fit into other C128 models as well. 

Today I bought a C128 for a good price. So I can try to design ONE FPGA board and maybe ONE adapter board for the C128 and the C128DCR. I check out flat flex cable - that would be a nice solution. 

When I do such a big step of "redesign" work it could be an advantage to switch to 0603 components. It is harder to solder, but it is better for a smaller design. I will see - maybe it is O.K. with 0805.

> Some possible ideas to shrink down the board: Use more of the buttom side. Indeed use a flat flex cable. You don't need the big heatsink for REG1 when you power the board with 5 volt.

I thought about improvements and I got similar ideas.

Well, I will start with this adaption to the C128, but it will take time. I still have other projects running. Learning KiCad will take time, but I will work on it. When I have some electronic questions can I ask you?
Shall I use this issues board, or is there another communication possibility?

c0pperdragon, 2021-01-16:
Hi!
I just had a longer walk through our current winter-wonderland here in Vienna  all the wile considering the C128 FPGA board.

As I have to wait for parts to arrive for another project anyway, I just can not resist to try design a super-small FPGA board myself. I envision a size that is equal to the smaller RF modulator box that is in the C64C as well as all the C128s. 
Besides the above mentioned things to leave out, I can also leave out the GND connectors on the left and right sides. I guess it will still be really tricky, but somehow this challenges me.
Being able to also put ito an C64C this makes it much more useful overall.
I will stick to the ribbon cable, though to not break compatibility with the existing adapter designs. 

Maybe there are other commodore machines besides the C64C/C128 that use the same RF modulator. So this could lead to other applications as well.

JDet-74, 2021-01-16:
> Hi!
I just had a longer walk through our current winter-wonderland here in Vienna all the wile considering the C128 FPGA board.

That's very inspiring - I guess ;-) - deep breaths of cold winter air - blowing through the gray brain cells :-)

>  I just can not resist to try design a super-small FPGA board myself. I envision a size that is equal to the smaller RF modulator box that is in the C64C as well as all the C128s.

Sounds good - so I did a good job pushing you in this direction. :-)

> I guess it will still be really tricky, but somehow this challenges me. Being able to also put ito an C64C this makes it much more useful overall.

Yes, that is right.

> I will stick to the ribbon cable, though to not break compatibility with the existing adapter designs.

I check my C128DCR if a ribbon cable will fit without collision with the metal plate of the PSU above.

> Maybe there are other commodore machines besides the C64C/C128 that use the same RF modulator. So this could lead to other applications as well.

I agree - maybe there are more applications and possibilities with such a small FPGA board.

If it helps you, I will take some detail photos of the RF-Modulator zone and the VIC area of both machines (C128 / C128DCR) and put the measurements into the pictures. That would help you by designing the FPGA and the VIC-adapter PCBs. I don't mind that.

c0pperdragon, 2021-01-17:
I actually managed to squeeze the board down to the size of the small RF modulator. I left out everything that is not necessary for this single-purpose application. Also left out the electrolytic capacitor which I had on the C64 board. It should also work safely without that.
Schematics and gerber files are located at: https://github.com/c0pperdragon/C64-Video-Enhancement/tree/master/smallreplacement

It would be nice, if you could do your trick with the cardboard templates also with this board to see how well it will fit.

JDet-74, 2021-01-17:
> I actually managed to squeeze the board down to the size of the small RF modulator. I left out everything that is not necessary for this single-purpose application. Also left out the electrolytic [...]

I will print out the new Version tomorrow and post the result immediately
Superfast work - dude - awesome :-).
When my C128 arrives, I will do the same cardboard technic again. So it sure, that it fits in both systems.
Will you also redesign the VIC adapter board for both fits?

c0pperdragon, 2021-01-17:
I don't want to redo the VIC-adapter yet again, especially when this would be of use for only very few machines. I really suggest you just relocate the capacitor which should be easy enough.

c0pperdragon, 2021-01-17:
Actually I have more than 10 VID-IIe adapter PCBs still. You can have 2 for the same conditions as @pazuzu72

JDet-74, 2021-01-17:
That looks very nice ... 

[image]

[image]

Thumbs UP :-)

c0pperdragon, 2021-01-17:
I hope it actually works. Design rules check and connectivity check are all OK.

JDet-74, 2021-01-18:
> I hope it actually works. Design rules check and connectivity check are all OK.

I will test it :-) 
When my C128 arrives, I will check the new small board on this motherboard.
By the way ... here are the pics of the small board in my C128DCR.
It looks still pretty.
[image]
[image]
[image]
[image]

It has to go with pin headers on the board, cause the ceramic caps (left, behind the DIN connectors) are maybe in the way.
But it fits very well. Here pics how it looks like in the metal case.

[image]
[image]

Next, I will check the VIC-adapter, if there is a possible solution for ONE VIC-adapter who fits in both machines.
If all looks fine, I order some prototype PCBs and the parts.
What do you think of this USB-Blaster? It should do the job.
[https://www.waveshare.com/product/fpga-tools/usb-blaster-v2.htm](url)
Can you recommend any developer boards? Maybe on the same "waveshare" homepage or any other boards.
You infected me with the FPGA virus.

Icelvlan88, 2021-01-18:
I’d still love to see RGsB one day :) I also use a mini din 9 for my connector. It works great

[FPGA_smallBoard_C128DCR_02]<https://user-images.githubusercontent.com/57826750/104875395-7aaa3780-5955-11eb-8897-62807f6ee511.jpg>
[FPGA_smallBoard_C128DCR_03]<https://user-images.githubusercontent.com/57826750/104875414-8695f980-5955-11eb-9255-46e7394ea9ad.jpg>
[FPGA_smallBoard_C128DCR_04]<https://user-images.githubusercontent.com/57826750/104875420-8a298080-5955-11eb-8df7-84dc0828f503.jpg>
[FPGA_smallBoard_C128DCR_05]<https://user-images.githubusercontent.com/57826750/104875432-91508e80-5955-11eb-9d44-034c485c8d08.jpg>

[FPGA_smallBoard_C128DCR_01]<https://user-images.githubusercontent.com/57826750/104875644-1176f400-5956-11eb-8730-5b28e3d9e2cb.jpg>
[FPGA_smallBoard_C128DCR_07]<https://user-images.githubusercontent.com/57826750/104875650-150a7b00-5956-11eb-989d-e65bd6e08d1a.jpg>

c0pperdragon, 2021-01-18:
RGsB  (a.k.a. sync-on-green) can directly be used by re-programming the palette. The palette-tuning software already support this.

To easily test if an adapter can fit both variants of the C128, just turn around the cardboard template and try to fit it with the connector on the other side. If this is clear of the capacitor in both cases, I think I will just redesign the adapter myself after all.

c0pperdragon, 2021-01-18:
> What do you think of this USB-Blaster? It should do the job.
> https://www.waveshare.com/product/fpga-tools/usb-blaster-v2.htm
> Can you recommend any developer boards? Maybe on the same "waveshare" homepage or any other boards.
> You infected me with the FPGA virus.

This USB-Blaster is probably just fine. 
About an FPGA development board:   I happen to have one of these lying around:

https://www.mouser.at/new/terasic-technologies/terasic-cyclone-v-gx-starter-kit/

I initially used it, but then moved on to building my own boards. So I don't have any use for it and I could actually make you a good price. Its also one of the few boards that have HDMI output capabilities built in.
Contact me via mail (reinhard.grafl@aon.at) if you are interested.

Icelvlan88, 2021-01-19:
Sorry what I meant was RGBs.

https://www.waveshare.com/product/fpga-tools/usb-blaster-v2.htm

https://www.mouser.at/new/terasic-technologies/terasic-cyclone-v-gx-starter-kit/

Contact me via mail (reinhard.grafl@aon.at<mailto:reinhard.grafl@aon.at>) if you are interested.

c0pperdragon, 2021-01-19:
@Icelvlan88  please refrain from quoting previous content in your posts if it is not necessary. This just clutters up the page.

For RGBS, you actually have several options:

1. Configure the palette to output RGsB and use the luma signal from the A/V port as sync. This works only with 240p/288p
If your monitor does not like having additional sync on the green line, you can even turn this off with the palette tuning software.

2. Configure output to RGsB and de-solder R1 of the FPGA board. Connect a wire to the R1s pad (the one directed towards the FPGA) , add a resistor (100 ohm or something like that) in series and that is your csync.

Icelvlan88, 2021-01-20:
is this TTL 5v or 75ohm if I put a 75ohm resistor if I do 2?

c0pperdragon, 2021-01-20:
Its more like TTL 3.3v, but this should also work with any TTL input.
When you use a 220 ohm resistor you also get voltages that are not dangerous to a 75 ohm input.

The sync inputs of monitors normally can accept a wide range of input voltages and a TTL input can often run with 75ohm sync and also the other way round.

So, if you just use a 220 ohm resistor, you should basically be compatible with everything.

JDet-74, 2021-01-22:
@c0pperdragon yesterday your package arrived. Thanks. The FPGA journey can begin ;)
The C128 arrived too and that are the results with the small FPGA board.
[image]
looks not bad - I don't want to be annoying, but if you just could please round the edges, it would fit better.
You will see it on the following pictures.
[image]
**WITH the C128 plastic case:**
[image]
[image]
[image]
The TRRA Audio Jack and the switch are fitting good.
[image]
If you just could squeeze the width of the board?! Left and right just 0.5mm each. Rounding the edges would be also better.
After this modification it will fit perfect, I think.

**The VIC adapter situation:**
I made a test adapter with the cardboard and a 32-Pin socket to play around.
[image]
If you reposition the 20-pin header to the VIC Pins (PIN-25 to PIN-35) / MOS 8701 clock generator, the adapter board would almost fit in both systems. It is very close, but the clock generator is in the way. A DIP-48 with longer pins should work an solve this Problem.
**C128 VIC-adapter, header on PIN25-35:**
[image]
[image]
**C128 DCR VIC-adapter, header on PIN25-35:**
[image]
[image]
The ideal position for the C128 DCR is like that (VIC PIN34-44):
[image]
But this doesn't work in the C128:
[image]
It's a mess - what's your opinion? A DIP 48 socket with longer pins and ONE solution for both models, or leave the adapter as it is and replacing the cap for the C128 DCR users?   
It would be possible to replace this cap with a 100µF tantalum cap like this one.
[https://www.reichelt.de/tantal-kondensator-rm-5-100-f-6-3v-tantal-100-6-3-p20336.html?&trstct=pol_8&nbc=1](url)

[61 more replies]

closed 2021-04-27 by c0pperdragon