c0pperdragon/C64-Video-Enhancement #5: Modify palette on the fly

chriz74x, 2019-03-21:
Could an internal api be added to modify the c64 palette on the fly from a bigger palette (16 or 24 bit) or maybe just use a different palette as you can in vice?

c0pperdragon, 2019-03-23:
This would probably mean to provide some registers in the VIC-II emulation. This would actually make the mod more powerful than the original VIC, so it would instantly disqualify itself for demo competitions.
I decided against this also because I have filled up the FPGA up to 95%, so I can not squeeze more functionality inside.

chriz74x, 2019-03-24:
Well it could be switched off and it would also create a league of it’s own for demo coding.

c0pperdragon, 2019-03-25:
Using a different palette altogether is possible when modifying the .VHDL source code and and using a custom firmware. A switchable palette is more problematic, because I just have no switches or jumpers on the board. Re-purposing the slider switch to switch between three palettes instead of the output mode would be somehow possible, but this again needs a custom firmware.

closed 2019-04-15 by c0pperdragon