c0pperdragon/C64-Video-Enhancement #6: BOM/parts list

bodgit, 2019-03-26:
I'm trying to put together a BOM/parts list to build my own board.

The IC's are simple enough to find by their part numbers but I was wondering if you could tell me what parts you used for the TRRS jack and the slide switch and/or where you purchased them from? I couldn't see any reference to a part number in the KiCad files.

Looking on Mouser, I found an SP3T slide switch (NKK SS14MDH2) that looks like it might be the right part (it's red and has the same markings) but I couldn't find something that looked like the same TRRS jack.

Is there any particular requirements for the the passives?

Sorry for all of the questions.

c0pperdragon, 2019-03-26:
Yes, the slider switch is exactly this one: https://www.mouser.at/ProductDetail/633-SS14MDH2

I intended to get all parts from mouser, but I just could not find a suitable TSSR jack there. I bought it from a local electronic dealer:  
https://www.conrad.at/de/conrad-components-klinken-steckverbinder-35-mm-buchse-einbau-horizontal-polzahl-4-stereo-schwarz-1-st-718732.html
I guess this is some generic part, this dealer sells under its own brand. But I don't know who the manufacturer actually is. Maybe you can somehow find this out?

The passives are pretty uncritical. I am using 1% tolerant resistors. Most parts are 0805 inch, with the exception of the 100nF caps that are 0603. These are that small to get a resonable low ESR. The rest should be pretty clear from the schematics and the photograph.

bodgit, 2019-03-27:
The TRRS jack looks like it might be a Cliff FC68125: http://www.cliffuk.co.uk/products/jacksockets/js354pole.htm

The photo, pinouts and data sheet measurements seem to mostly match up, some measurements are out by 0.1-0.2mm but there should be enough wiggle room on the PCB for it to fit.

c0pperdragon, 2019-03-27:
Yes, it looks like one part is a rip-off of the other, with nearly the same dimensions and the same pin assignments. It also has the same kind of springy pins.
So I guess the Cliff FC68125 will fit into the existing PCB just nicely. A shame that mouser does not have it, otherwise ordering would be much easier.

bodgit, 2019-03-27:
Looking on Mouser for a through-hole, 4 connector, non-switched female 3.5 mm jack produces only two results, one of which is this CUI part: https://www.mouser.co.uk/ProductDetail/CUI/SJ-43514

The PCB footprint would need tweaking for that to fit but you could in theory get everything from Mouser that way. I can find this Cliff part easily enough from a few places so it's not a massive problem.

c0pperdragon, 2019-03-27:
This part from mouser has a wider footprint, so you would need to re-arrange the PCB quite a bit. It is also lower, which means, you need a cut-out in the PCB because the barrel of the jack extends below its surface. And then the horizontal alignment of the jack and the slider switch will not match the holes in the computer case at the same time. This is especially the true for the C64C case where the slider switch is nearly too high already.
So, this part just doesn't work so well.

I tried to find other distributors from where to source everything in on go, but Digikey also does not have a comparable socket, and farnell has basically no FPGAs. So I had no choice but to get the parts where I found them.

bodgit, 2019-03-28:
I think I've put together a BOM that has almost everything: https://www.mouser.co.uk/ProjectManager/ProjectDetail.aspx?AccessID=05df40c7cd

The only parts I'm struggling with are the 3 inductors, what have you used here?

I think I have all of the resistors, I couldn't find 4k or 8k values exactly so I picked 4.02k and 8.06k, I presume they should be close enough?

Apart from the 100uF electrolytic capacitor, I've used X5R ceramics everywhere else, is that ok? Should any of them be explicitly low ESR or anything like that? All of the capacitors are rated for 16V as I figure there's 12V supply involved.

c0pperdragon, 2019-03-28:
The inductors are just the cheapest 10uH inductors in a 0508 form factor I could get. I have no real knowledge of their electric properties anyway, I just hope they will help with filtering the switching noise. I have found this arrangement in a application note somewhere and it seems to work for me. 

I also used resistors that are slightly off, this works just fine for my DAC. 

As for capacitors I also have no real clue about that the different types are for. I just pick the cheapest of the given capacitance and form factor.  Yes, 16V is good, at least for the ones sitting before the voltage regulators. But of course you would not buy different types, so use just 16V everywhere.

bwack, 2019-05-07:
@bodgit Thanks for sharing the parts list. Have you got the parts by now ? I a wondering if the Cliff FC68125 TRRS jack socket fits the footprint ?

desaster, 2019-05-07:
> @bodgit Thanks for sharing the parts list. Have you got the parts by now ? I a wondering if the Cliff FC68125 TRRS jack socket fits the footprint ?

I've just built the board, and can confirm that at least this FC68125 was a good fit

https://www.ebay.com/itm/2-X-FC68125-Socket-Jack-3-5mm-female-ways-4-THT-W-7-1mm-H-6mm-D-15-1mm/113687766045

bodgit, 2019-05-07:
@bwack I've got the parts but I've not yet built the board, (still plucking up courage to try and solder the FPGA QFP, any tips?). I did do a dry fit of the Cliff jack and it seems to fit the holes in the PCB no problem, I got them from Rapid Electronics.

bwack, 2019-05-07:
@desaster Thanks for the link. I've ordered from that link now.

@bodgit I'm going to try soldering the FPGA with drag-soldering technique.

c0pperdragon, 2019-05-08:
Soldering these low-pitch parts is indeed quite tricky. My technique requires a magnifier (one of those combined with a  lamp) and 0.5 mm leaded solder and a not-too-fine soldering tip:
First bring a bit of solder on one of the pads next to a corner. You can place the lead of the part on this pad while keeping it hot with the iron and fine-adjust it while checking the alignment (I could not do that without help of the magnifier). Then solder a pin on the opposite corner again while fine-adjusting everything.
Soldering the rest takes quite some time and will probably lead to some solder bridges. Use solder wick to remove these. Always apply plenty of flux.  Good look.

bwack, 2019-05-29:
I'm going through the bom list, checking footprints.. I noticed that the 40 pin precision socket was different than the one listed in https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod (where it says "Vic Adapter Board.....I have set up a parts list at mouser") than in the mouser parts bom list posted above here by @bodgit . It seems that the pins are longer in the one @bodgit ordered/listed. The reason why I'm asking is that I've accidentally bought 1.6mm pcbs instead of 0.8mm VIC adapter boards. Can the long pin ones work with 1.6mm maybe ?

bodgit, 2019-05-29:
@bwack the 40 pin socket I chose has plenty of length in the pins, you should still be fine with a thicker PCB. I'll see if I can take a photo later if that helps.

I wasn't sure if I picked one with far too long pins as I didn't have a reference at the time. In my head I thought I'd ordered a wirewrap-style socket with the really long pins!

c0pperdragon, 2019-05-30:
That is very nice! I didn't know that sockets with such long tails actually existed. They are a bit expensive, but it is well worth it to reduce the potential problem of poor contact with my original design (which was in fact the only thing with which I was not very comfortable).
I would really like to hear how this works out with your 1.6mm board. As the pins are 1.1mm longer than on standard sockets, this should more than compensate the 0.8mm of increased board thickness.

bwack, 2019-06-08:
@bodgit 
> I think I've put together a BOM that has almost everything: https://www.mouser.co.uk/ProjectManager/ProjectDetail.aspx?AccessID=05df40c7cd

Hi Matt. Oops there is a part with wrong pitch in the mouser-bom, Mouser No. 855-M50-3801042 "Headers & Wire Housings 10+10 DIL IDC ML, Au TRANSITION PLUG" is 1.27mm pitch, but the footprint on the board is 2.56mm pitch. will just solder the ribbon cable directly to the board in my case.

Alternative: https://no.mouser.com/ProductDetail/TE-Connectivity-AMP/1658525-9?qs=sGAEpiMZZMs%252BGHln7q6pm48SVpWlpfsEIgMEoMh%2FhDU%3D Can you update the bom, please ?

bodgit, 2019-06-10:
@bwack Good spot! I don't think I noticed as 1.27mm is the pitch of the IDC ribbon cable and I had some of those connectors left over from another project so I added it to the BOM but didn't order them myself in the end.

BOM updated with your suggested part.

bodgit, 2019-06-12:
What's the recommended way to solder that ground pad on the bottom of the FPGA? I've practiced with some other QFP chips and it's actually quite straightforward to do the legs, easier than I thought anyway! The hardest part is getting the chip lined up with the pads on all four sides. I'm curious if it's best to plate the PCB underneath the chip first or try and do it through the hole from the other side. What technique did you use?

c0pperdragon, 2019-06-12:
The ground pad is actually the easiest part.
Once all legs are done, just turn up your soldering iron a bit and feed liquid solder into the hole. The hole has metallic sides that are directly connected to GND. As the ground plane drains away the heat very quickly, the solder will turn solid after it has made first contact to this metal. Keep heating the solder until it eventually becomes liquid again. Use your soldering iron to stir the solder inside the hole until no solid junks remain  (like when defrosting some vegatables in a cooking pot). Then it is done and the solder will have made very good contact with the ground pad as well as the sides of the hole.

jannekayra, 2020-01-09:
Very interesting project indeed and I'm most likely going to attempt building it.

I am looking at the BOM at Mouser and noticed that the IDC male transition connector for the VIC adapter board is quite expensive. Is there any reason to not use a right angle IDC male socket instead?

c0pperdragon, 2020-01-11:
I wanted to keep the construction as low as possible, so it will not hinder the use of heat sinks on the VIC. Or cause any problems in a C64C case.

But if you are building your own kit, you will best know the constraints of your installation. If you have enough space you can even use straight pin headers (did work for my first breadbin prototype installation:  https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod)

mikketeus, 2020-03-23:
Is thick film resistors necessary or should it work with regular SMD resistors as well?

c0pperdragon, 2020-03-23:
You can basically use any type of passives for the build and there is not much demand for precision. There is one exception though: Try to get resistors for the DAC networks reasonably close to the respective power-of-two values. 

I guess for SMD purposes you do not have much choice and you can basically only use thick film or thin film resistors. Thick film resistors seem to be a bit cheaper and more common, so I use those for everything.

closed 2020-07-30 by c0pperdragon