c0pperdragon/C64-Video-Enhancement #9: Problem with corrupted video output / RAM issue?

desaster, 2019-05-08:
Hi,

I've built the mod, and got it almost working. Attached is a picture showing the type of corrupted image I'm getting.

I'm still working on troubleshooting as much as I can myself, but I thought i'd post this issue meanwhile to get some ideas on what to check.

## My hardware

* PAL motherboard #250425
* VIC chips tested:
  * 6569R3
  * 6569R4

## Things I've done
* Connected the board to c64 with a temporary connection via GND3 and RFCON2
* Checked my soldering work for shorts or unconnected pins
* Built a second VIC-II adapter board, since I wasn't sure how to properly test the LVC257

## Things I've not yet done
* Tested the FPGA chip, my plan is to write some verilog to make sure the pins are working correctly
* Double or triple checked all connectivity

Below are the comparison pictures. In C64 startup screen there are lines going next to each column. In games, sprites also seem to have some issues. Sorry about the bad pictures!

My wild guess is just that some of the data or addressing lines are not working. But maybe it could be some VIC compatibility issue too?

[image]

bwack, 2019-05-08:
Is one of the data lines stuck at one ?

c0pperdragon, 2019-05-09:
When you have built a second VIC-adapter already and this is the same picture, than I can basically only by a problem with one of the data lines. D7 as it looks like. This would be pin 74 on the FPGA. Try to resolder this one.

desaster, 2019-05-09:
Thanks guys, you were both right. The pin 74 was shorted to +3V3.

This is a bit embarrassing, because I had gone through all the pins looking for shorts, but obviously didn't do a very good job!

The picture looks great, and I'm now ready to do a permanent install.

[image]

closed 2019-05-09 by desaster