c0pperdragon/C64-Video-Enhancement #11: Giana Sisters II glitchy time display

desaster, 2019-06-02:
Hi,

I noticed a small issue with the time display in Giana Sisters II. After the game starts, if you walk forward for a bit to cause the screen to scroll, the time countdown starts having some glitchy lines flickering under the digits.

I tested with the latest 2.0 release, as well as the "C64mod_2_1_alpha2.pof" experimental firmware.

To make troubleshooting easier, a trainer version of the game can be used to prevent the time running out, while still displaying the glitch.

PS. Do you prefer a separate issue ticket like this?

[image]

c0pperdragon, 2019-06-02:
Thank you for this bug report. And yes, please just open new issue threads for individual problems.
This one now locks like a problem with the exact timing of when the xscroll register should actually take effect. The program obviously changes the xcroll value at the end of the score bar to then have it correct for the game area. I am not sure what the correct behaviour should be. Maybe I find some test programs in the VICE test bench to figure this out.

c0pperdragon, 2019-06-04:
I have again spent several hours to improve the VIC emulation. Please feel free to get a new experimental firmware that should fix this glitch and probably a few more from 
 https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/experimental

It would be nice if you could test this with the "Brutality" demo also. This program is especially hard to test for me, because it quickly crashes my computer (a known problem with the DMA delay technique).

desaster, 2019-06-05:
This version seems to fix the glitchy time display, but unfortunately seems to introduce some new issue with sprites.

I noticed this first at second level, as seen in the screenshot below. However, I think the broken sprites may be easier found elsewhere. I only have time for quick tests right now, but was able to see the same issue with the EdgeOfDisgrace demo.

[image]

[image]

c0pperdragon, 2019-06-07:
I could not reproduce this issue on my machine, but I have changed the sprite handling for other reasons anyway. So maybe the new firmware 2.1 (availabe in releases secion) fixes some of this problems.

desaster, 2019-06-08:
Here are some videos of firmware 2.1. Whenever the audio cuts out, that's when I'm switching to normal A/V output to check what the original looks like.

* [Giana Sisters II](https://www.youtube.com/watch?v=_25I0TWcqrA) - glitchy sprite can be seen at around 1:12
* [Brutality](https://www.youtube.com/watch?v=ajVIkzhPvgE) - problems with left border (0:28) and sprites (1:56)
* [Edge of Disgrace](https://www.youtube.com/watch?v=kwQ5qdU4D78) - various cases of border and sprite issues

desaster, 2019-06-08:
more small that show the sprite issues:

https://csdb.dk/release/?id=166859
https://csdb.dk/release/?id=159699
https://csdb.dk/release/?id=159706

c0pperdragon, 2019-06-08:
All your examples (with the exception of Edge of Discrace, which I can not yet test) work flawlessly on my machine with firmware 2.1   So I very much guess that it is a problem with one particular information I receive from the VIC which has a very small stable time. It seems that your machine has this timing just a bit different than mine. To try this out, I have shifted this particual sampling point back by about 30 nanoseconds. This still works on my machine with a bit of additional tolerance. Maybe this fixes all this problems in one go.

Get this experimental tweak from:
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/experimental

desaster, 2019-06-08:
with 2.1p1 the sprites are fixed!

Tested:
* The three little sprite demos - all stable
* Giana Sisters II - sprites are fixed, and of time display is still ok
* Brutality - borders looks good in the scroller, and no sprite issues
* Edge of Disgrace - sprites are fixed, but border related issues still remain (just like in my linked video minus the sprite issues)

c0pperdragon, 2019-06-08:
So next I will concentrate on getting Edige Of Disgrace working.

closed 2019-06-08 by c0pperdragon