c0pperdragon/C64-Video-Enhancement #13: Thermal releifs on pad connections

bwack, 2019-06-08:
Hi. The decoupling caps under the board were hard to solder on the large copper fill. Can you add thermal releif ?

[image]

C64 Video Enhancement board rev 3

c0pperdragon, 2019-06-08:
I decided against thermal reliefs to make the the decoupling caps as effective as any possible. Soldering the caps is easier if you take care to first solder the side that is not on the ground plane. Once the cap is firmly attached, the you can heat up the other side for a longe time without trouble.

bwack, 2019-06-08:
Thanks for the tip!

closed 2019-06-09 by c0pperdragon