c0pperdragon/C64-Video-Enhancement #14: Trouble shooting

bwack, 2019-06-15:
Hi. Thanks @desaster for recommending to use the test pattern. I can see a pattern on my scope, but not on the TV. It seems to be very slow. Maybe that is it.
The sawtooth sweeps looks fine, so I guess the resistor ladder is ok.
[image]

Maybe the fpga needs configuring.. ?

desaster, 2019-06-15:
when powering the board, notice that there are two groundplanes, GND and GNDA. GND1 and GND2 are connected to GNDA. I initially made the mistake of trying to power the board with GNDA instead of GND.

bwack, 2019-06-15:
Ok. I did that mistake the first time. Now the board is in the C64. I see I have soldered four grounds except GND5. Not sure what to do. Maybe the crystal is wrong.

bwack, 2019-06-15:
Ouf, I have mounted the crystal wrong..
[image]

[image]

bwack, 2019-06-15:
It was the crystal. Now there is picture :) Thanks for help !

c0pperdragon, 2019-06-16:
Funny thing that the FPGA actually generates a signal without any oscillator at all. It seems that the PLL inside falls back to generating an ultra-low-speed clock (more than 1000 times slower as planned) when not being driven by an external signal.

closed 2019-06-17 by c0pperdragon

bwack, 2019-06-23:
Maybe the VCO in the PLL continues without the lock.. or there is a fallback. I tried to figure out by datasheet, but without luck. Just curious anyway :)