c0pperdragon/C64-Video-Enhancement #16: Let's Invade! - Corrupted text and stars in main menu

desaster, 2019-07-03:
Hi,

I've been testing Firmware 2.3 a bit, and just stumbled on this issue with "Let's Invade!". 

Links:
https://tnd64.blogspot.com/2016/10/run-run-invasion-has-started.html
https://csdb.dk/release/?id=151230

Picture comparision:

[image]

With the FPGA implementation, there seems to be corruption in the main menu texts, as well as the starry background. The rest of the game seems to have no errors.

desaster, 2019-07-03:
This is probably related, since there's something funny going on with text. In a finnish game "Alien Bash", the text is also slightly corrupted in the main menu (there seems to be extra pixels). 

Links:
http://www.ntrautanen.fi/computers/commodore/archive/c64_pelit.htm
http://www.ntrautanen.fi/computers/commodore/archive/c64/pelit/alien_bash.prg

c0pperdragon, 2019-07-03:
I have broken the multi color text mode in the last release, so I just provided a bugfix version on the releases page.

desaster, 2019-07-03:
tested and confirmed, all fixed!

closed 2019-07-03 by desaster

desaster, 2019-07-03:
Although with 2.4 I'm now experiencing some major issues with brutality / edgeofdisgrace. I'll make a ticket tomorrow.