c0pperdragon/C64-Video-Enhancement #17: Screengarbage and load errors

bwack, 2019-07-03:
When the vic ii is in the adapter, and i use jiffydos or other fastloader the screen fills with symbols and sometimes a load error appear. I have checked the socket and it seems fine. Swapped cias no change. I disconnect ribbon cable to the fpga board and no change. Inserting the vic without adapter no problem.

bwack, 2019-07-03:
Ok now i have the issue without the adapter. Ill have to repair my computer before i continue.

bwack, 2019-07-03:
Removed the swinsid ultimate and problem resolved. No idea..

closed 2019-07-03 by bwack

c0pperdragon, 2019-07-04:
The VIC adapter board is designed to only disturb the rest of the system as little as possible. Each signal is sensed with one high-impedance input that should draw less than 1 microampere and has an input capacitance of about 5 picofarad. 
But of course when the rest of the system is already on the very edge of stability this tiny difference may actually push it over the edge.

bwack, 2019-07-04:
Yes. Could have been the drop that filled the bucket. Anyway the computer works fine without the Swinsid Ultimate. I dont think the fault is related. I was concerned that maybe flux residu could cause problems on thr 0.5mm pitch pin, but no..

desaster, 2019-07-04:
#18 is probably not related, since you were also having issues with A/V out.

bwack, 2019-07-04:
@desater. Yes, this happened on my computer with and without the vic adapter. No problems after removing the swinsid ultimate.
The early SSU that I have is covered in hot-snot glue. I've heard about glue becoming conductive after a while, but that is just a guess. The glue covers the system bus lines on the SSU board.

DeSater, 2019-07-04:
@bwack I think you mis-tagged