c0pperdragon/C64-Video-Enhancement #18: Random video corruption / sync loss

desaster, 2019-07-04:
Hi,

After 2.4 I begun experiencing some random screen corruption.

I've reproduced this by playing the Edge of Disgrace demo a few times. Half the time there are no issues, but then sometimes during the demo I experience several sync losses followed by corrupted screen.

When the corruption occurs, the A/V output still displays the image with no problems.

I've uploaded a video on youtube of a full demo playback.  Corruptions are at several places, jump to [4:30](https://youtu.be/z8sQeGt_z_E?t=270) for example.

https://youtu.be/z8sQeGt_z_E

This is a bit hard to reproduce. To narrow this down to any specific firmware change, I'll have to run a lot of systematic tests while downgrading the firmware gradually.

While this bug report is against 2.4, I cannot be sure if earlier versions also had the issue. I've had some strange sync losses with my OSSC - but I didn't notice screen corruption then.

The sync loss with corrupted screen (loss of state?) seems to me like the whole device might be resetting.

[image]

c0pperdragon, 2019-07-04:
After watching this video, it seems to me that the FPGA loses the sync to the VIC. So it then completely miss-interprets much of the signals. This would also explain the screen outages, as csync signal generation by the FPGA jumps to a different state, which the TV needs some time to recover from.

The way the FPGA syncs itself to the VIC in the first place is a bit tricky. I try to detect the access pattern for the RAM refresh and from this I can deduce when the first line of the screen is generated and in which cycle the VIC currently is. Unluckily this access address is stable for only a very short time and I have to sample it at just the right moment to get reliably reading.
It can very well be that depending on temperature or the relative harmonic state of the clock generation chip to the internal counter in the VIC this now started to fail on your machine.

I will try to experiment how far the sample point can be moved forward and backward while still working stable on my machine and try to place it then right in the middle of this interval. This is basically the same thing I had to do for the sprite issue you reported in https://github.com/c0pperdragon/C64-Video-Enhancement/issues/11  and also for the incompatibility with the Ultimate cartridge you reported as well.

All this hints to your specific machine really being the worst case in regard of signal timing issues. So if we get it to work there reliably, this should basically cover all other machines as well.
And I don't think the older firmware will behave any better in this respect.

c0pperdragon, 2019-07-05:
At some experiments I found that the sample point for this data can be moved around on my machine in an interval of 125ns. I have built an experimental version where I moved this point backwards by about 30ns  (exactly as I did for the sprite issue). Maybe this improves the matter on your machine.

https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/experimental

desaster, 2019-07-05:
Thanks, I've applied the firmware and will start testing. I don't know how soon we can confirm if this helps, since the issue has been very intermittent.

desaster, 2019-07-08:
So far the experimental firmware has been working flawlessly.

c0pperdragon, 2019-07-08:
Thank you, that is good to know. I will make this version now the official 2.5 .

closed 2019-07-08 by c0pperdragon