c0pperdragon/C64-Video-Enhancement #19: Great picture quality! Some questions regarding color(s).

olliraa, 2019-09-09:
Just installed the mod finally yesterday, and I must say the picture quality on my Sony BVM20F1E is stunning, just incredible :D

One thing I noticed is that the colors are a bit "off" compared to what I've used to with y/c on the same monitor. Changing the color setting from RGB to YUV in the monitor OSD helps a lot, but still for example the lila/grey/blueish default border color in the "boot screen" is a bit more turquoise now. Also all colors seem to have more "tint" , which of course looks very sharp also. This color change can also be noticed from the example screenshot of World Games -> the character shirt is way more blue than any other screenshot in the web.

This is not to complain, but I'm curious is this just the effect of all the official ways of displaying the picture just so "washed off" compared to this mod?

c0pperdragon, 2019-09-09:
Very good! I am glad this works so well for your setup.

About the colors: YUV is similar to YPbPr, but not quite the same, so you can expect to have some differences. Also the default color settings are quite saturated, because I wanted to give the picture a fresher, less washed-out look.

Anyway: Beginning with firmware 2.0, I have implemented a system with which you can tweak every color to better fit your liking.
Explanation for this is at the bottom of the main project page (https://github.com/c0pperdragon/C64-Video-Enhancement).

There is actually also a user-developed program with visual interface to help you with this:
https://github.com/Hojo-Norem/c0pperdragon-VIC-II-Palette-editor

olliraa, 2019-09-09:
That palette tuning software is awesome. I just selected new lumas and the colors are now "correct" :D I have a 6569r5 on 250466 mobo. I really do not know whether the new lumas should do the trick, but apparently they did. Woo-hoo :D Thank you for this awesome mod for the best computer of all times!

closed 2019-12-17 by c0pperdragon