c0pperdragon/C64-Video-Enhancement #24: User feedback and experiences (Part II)

c0pperdragon, 2020-03-14:
Please let the community know about your experiences when building/installing/using this mod.
This is a continuation of https://github.com/c0pperdragon/C64-Video-Enhancement/issues/7 .

Icelvlan88, 2020-04-05:
Build an RF replacement and A-video board. All I'm getting after installation is a black screen. Used the C64mod_2_6_1 firmware. Not getting video out of composite either, again just a black screen sometimes it is colour.

c0pperdragon, 2020-04-05:
I'm a bit confused. Did you build a RF replacement board, or an A-Video board?
Or a RF replacment and the 64 adapter (taken from the A-Video board repository).
It would be best to just add some pictures of your installation...

Icelvlan88, 2020-04-05:
I got a bit mixed up at first. So I built the RF replacement. I believe my issue is with the system itself. Before I started everything was working, but now all I get is a black or pink screen. Would it be possible to help me troubleshoot this a bit? Maybe by email is better since it's not really relevant here.
[image]
[image]
[image]

c0pperdragon, 2020-04-06:
I would like to keep this issue thread for communication. Emails are so messy, and a public thread is so much better to help others solving their own problems as well.

Now even having your pictures does not really clear up the issue for me. The second pictures seems to show how the desoldered RF modulator is still attached to the board via wires. No mod installed yet. Did this still work, producing a the signals on the A/V port?

The third picture shows the RF replacement board and the adapter board connected together, but not connected to anything else. On first glance, these two boards seem to look OK. I can not quite make out, what this strange socket is you have connected the video output signals to. Some kind of VGA-style connector, maybe?

Anyway I would like to see how you installed the mod into the C64 to give some more educated guess on what is amiss.

But you actually could also do some tests on the mod board alone to check if the FPGA itself and the video output circuit is working (and your monitor setup as well).
This firmware:  https://github.com/c0pperdragon/A-VideoBoard/releases
just generates a 576p test pattern, without using any of the input lines, so it will show an image on all screens that can accept this type of signal (basically all TVs that have a YPbPr input).

c0pperdragon, 2020-04-06:
OK, I had a second look on the third picture and no I see that you had the board installed in the C64 already (seeing some solder residue on the RFCON1 pins) . So I guess this did not work then and you unsoldered it again and then using the original RF modulator it still does not work.

I really hope that you did not damage your C64 in the process. There are two things I could imagine causing damage: 
1. Shorting some of the input signals of the adapter board with solder bridges. This would lead to the CPU / RAM / ROM  / VIC being shorted too causing all types of stress and damage.
2. No proper contact between the adapter board and the main board or the adapter board and the VIC. This may lead to bus conflict in the unlikely case that the VIC now longer properly disables the CPU in the relevant situations.

I fear I can not be of great help in troubleshooting a really damaged C64. I just happen to have a damaged machine (a C64C) myself and just can not find out what is the problem there.

Icelvlan88, 2020-04-07:
Looks like I had a bad PLA.

Another questions. Is it possible to add so component and RGB outputs at the same time? If not, is it possible to add a switch that would allow me to switch between both?

Hojo-Norem, 2020-04-07:
You can't output RGB and component at the same time but you can switch between them using the palette editor.  Get the palette just how you like it in component mode, save it to tape or disk, switch to RGsB (or RGBns if you make a sync cable), tweak the settings if needed and then save that to tape or disk.  Whenever you want to switch between RGB and YPbPr all need to do is load the relevant palette file.  You don't even need to load them into the editor first as the palette files can be loaded straight from BASIC.

Icelvlan88, 2020-04-07:
I’m assuming you mean RGBs? Is it TTL or 75ohm sync?

Hojo-Norem, 2020-04-07:
Nope, RGsB, aka RGB with sync on green.  'RGBns' is RGB without sync, which means you have to use the luma signal from the C64's AV port.  TTL or 75ohm sync?  I don't know, I'm not really an expert.  I suppose the sync level is whatever YPbPr uses as that's how c0pperdragon designed it.

Icelvlan88, 2020-04-07:
Might be component sync as well then. I want to try and put a mini din connector instead of a TRRS.

c0pperdragon, 2020-04-07:
The original A/V port provides a luma signal (greyscale with embedded sync pulses) as well as composite video (brightness, color and sync pulses all combined to one signal). You could use either one as a source for sync in the case that you take pure RGB from the mod board. 
But this is neither a TTL sync nor a "pure" 75 ohm sync. Whether your monitor can use this as sync input, you really have to try. 
Monitors with a SCART input can normally be made to work with that, as the SCART plug carries wires for R, G, B and component video.

Icelvlan88, 2020-04-08:
I'll have to make a custom cable and find out. Are there other features in the works?

rapiqui, 2020-04-12:
Hi!

I'm currently in the process of building one C64 Video Enhancement RGB board.
I have encountered some difficulties (well, I guess that might be normal if you're not too versed on SMD soldering and FPGA, such as ruining an Altera MAX 10 because of bad allignement), but other than that the project seems to advance.

I currently lack some components (proper oscillator, some resistors, inductances and capacitors), but I had in mind to program the Altera FPGA when discovered that the MAX 10 is not supported by Quartus II 12.0sp1.

This is how it currently looks like:

[image]

I will let you know once I could finish it, program and test it on a C64 250407 with a 6569R5 VIC-II and an Internal YPbPr to HDMI converter

Icelvlan88, 2020-04-12:
I used Intel FPGA

[IMG_20200412_164850]<https://user-images.githubusercontent.com/55879300/79073260-d4564500-7ce5-11ea-812a-9b60be50c56a.jpg>

c0pperdragon, 2020-04-14:
As far as I see, your soldering looks reasonable. Do not forget to solder the ground pad of the FPGA from the backside.

Icelvlan88, 2020-04-15:
I beielve the board needs to be powered/ground by c64 to program. I think there was reference somewhere of the points

c0pperdragon, 2020-04-15:
You can program the board outside the C64, but you need to provide proper power on the correct pins. Look at this for more details:
https://github.com/c0pperdragon/C64-Video-Enhancement#create-your-own-kit

rapiqui, 2020-04-15:
> I beleive the board needs to be powered/ground by c64 to program. I think there was reference somewhere of the points

That was clear in my case, and I did soldered the corresponding VCC and GND pins on the C64 motherboard, however I am lacking the crystal oscillator and the inductances, hence I think I won't be able to program it (or Quartus to detect the FPGA) until this is finished.

Icelvlan88, 2020-04-21:
Any chance this FPGA is able to push a higher resolution? 720p?

c0pperdragon, 2020-04-22:
This FPGA ist nearly at its limits  regarding the amount of functionality. But there is quite some room for higher speed. Would require considerable Code adjustment, though. Also it gets increasingly difficult to produce an analog signal with higher bandwidth. Especially the video output amplifier is designed with a filter that may soften the output.  
One question: What would be the use case for analog HD ?

Icelvlan88, 2020-04-22:
Was just asking, not an issue at all. Would be nice to use in a TV especially in 1080p or close, then we don’t have to use the scaler in the TV.

Hojo-Norem, 2020-04-22:
720p would be NTSC 240p line-tripled.  PAL C64 base resolution is 288p and tripling that gives you 864p.  I guess just doing 3x lines would be easy to do, but I don't know how compatible 864p would be.

To do PAL 720p I guess that you would need some line buffering and scaling on the FPGA.  I guess it would need a bigger FPGA (I don't suppose there's one that's pin compatible with the current one?).  Also it would be better to target 1080p and implement some form of hybrid integer+bilinear scaling.

Still, an OSSC or even better the upcoming OSSC Pro (it supposed to have a <1 frame lag scaler) should work well with the mod as it is.

Icelvlan88, 2020-04-22:
The good one to have is 3X and 4.5X. Yes the OSSC pro would help.

MagerValp, 2020-04-22:
Can confirm that it currently works great together with the OSSC.

TMS2764, 2020-05-01:
Its working great for me. Here a few things I realized along the way.

The 3.5 mm video adapter cable I was using did not have the right configuration. The signal ground was not on the sleeve ring but on the second ring. I still got an image but is was bad. Until I realized the cable configuration was the problem, I thought the board build might have something wrong.

Finding the right stuff. The VIC adapter board is in another project "A-VideoBoard/c64mod/". I could not find the programming file so I ended up compiling what was in "A-VideoBoard/c64mod/quartus/" with core updates and timing closure issues which I did not know if it was normal. Then I read in the comments that the programming files were in the "releases" tab. Grrr! I would have appreciated some links or a table of where everything is for this project.

Thanks for sharing this project. It was fun to build.

[image]

[image]

[image]

c0pperdragon, 2020-05-01:
Very nice build and installation.  You have quite some substitute hardware on your board already...

rapiqui, 2020-06-25:
> That was clear in my case, and I did soldered the corresponding VCC and GND pins on the C64 motherboard, however I am lacking the crystal oscillator and the inductances, hence I think I won't be [...]

Update 25/06 - The 4.02K resistors arrived and installed them in the board, the inductances haven't arrived yet.
I have tried to program the board with Quartus Prime using the c64mod_2_6.pof file, but Quartus is unable to communicate with the Altera Max 10 FPGA, although the board has been powered by placing it on the C64 (and I was able to measure 3.3V from the output of the board regulator).

So, now I doublt if the inductances required in order that Quartus could communicate with the the Altera MAX 10 or perhaps I could be suffering a chinese ST based Altera USB Blaster programmer.

Regards

c0pperdragon, 2020-06-25:
Maybe you did not properly connect the power. This can easily happen, because the mod board has two different ground planes and also two independent voltage regulators: One for the digital signal path and one for the analog path to drive the original A/V connector.
So you basically need to connect one of the pins GND3,GND4,GND5 to the main board ground and the rightmost hole of RFCON2 to +5V.
This should be enough to get the digital part of the mod to work - and also enough to programm the FPGA. 
Also take care to attach the programmer cable in the correct orientation (may seem a bit counter-intuitive)

rapiqui, 2020-06-25:
Hi. Thanks for poiting it out so quickly :)
I have GND1 currently connected and the header of RFCON1 and RFCON2 fully populated with all four pins in each header.
I had previously tested continuity between GND3 pad and GND1 pinheader (soldered on the C64) and there was continuity.

USB Blaster connector pin 1 is next to the REG1.

The 3,3V i measured were on the REG2 (did not test it on the REG1), but I guess that one will be for digital and the other for the analog part.

c0pperdragon, 2020-06-25:
You absolutely need to connecta at least one of GND3,GND4,GND5  to the board ground. I don't know how you got continuity between GND1 and GND3. But even if there is some secret connection around somewhere this is not advisable for noise free operation. If you actually have GND1 soldered to the C64, you very likely have "long" board. In this case, please solder GND2,GND3,GND4 also to the board for best performance.

The Regulator REG2 is in fact for the analog part and is totally unconnected from the FPGA side. If you do not believe me that proper grounding is essential, please measure the voltage of REG1 and the voltage between of GND3 and GND1.

rapiqui, 2020-06-25:
Hi,

Apologies, I wrote GND1 when I meant GND4.
In any case you're right, this is a long board, so I will proceed to solder GND2 and GND3 too.

I have measured the voltage on REG1 and it was almost 0v!. It seems that the GND4 was not making proper contact.
I will solder the rest of pins and retry.

Thanks for your help! Fingers crossed on the FPGA programming.

pazuzu72, 2020-06-25:
I am using your upgrade with great pleasure on my C64 connected to an RGB CRT monitor. Up to now I have not found any incompatible programs or games ... today it happened! This is SdBrowse, not that I care, but maybe it can be useful to find a solution and understand the reason. Switching to composite obviously shows well.

https://csdb.dk/release/?id=152470

c0pperdragon, 2020-06-25:
Could you please open a new issue for this incompatibility. Best would be with some photographs as well...

lfitzsimmons, 2020-06-29:
Hi There ! Thanks for putting this together , Im having a fun time making this even though I have an Issue - wonder could you help?
RF Replacement Board went together OK,  when programmed with the test pattern I see it correctly (Im using OSSC) However with main program I see a garbled screen like this once connected with the VIC adapter board: 
[image]
Cycling the power sometimes changes the location of the screen elements, or occasionally nothing at all. Any Ideas? 
Thanks in advance.

c0pperdragon, 2020-06-29:
It seems that the greater part of the FPGA board works, including all of the output circuit.
Guessing from your image I would say that there are at least two bits of the input datapath not working correctly. One is obviously D6, but D0 seems also to have a problem, as well as one of A0 or A1.  

The problem may either be on the adapter board or on the FPGA board. Try to probe the signals comming through the connector cable and if they look plausible, try to check continuity and shorts on the FPGA pins themselves.
As you have built this together yourself, you probalby already know the schematics for both boards.
And I assume you have an oscilloscope. Otherweise troubleshooting would be pretty impossible, of course ;-)

lfitzsimmons, 2020-06-30:
Thanks for the quick reply - Yes defo need to follow the Schematics for this one :) I followed D0 all the way to the FPGA- seemed ok but saw some residue and general fluff on the input pins. Cleaning those up got me a little further: 
![3Y7iyQZ3zH-K2t9UYoVZNiTsJQ1UhyhyDQz8aRLnD--fHo-J4guugig33Og5NaB-3_hBgm8pYOcnyDTysZkLg997T8enZdHPrKox1DCLt2AYQ0LqYIyIPHytBAeZg6auVDg9Bsrf4sYr3ago27-z7GmTjQsz2o8Lk_fCo4dnxnyF_purZ2NdKRvaUfe0uCJJk5_S59rV4nYK2sa4Xa-Cy](https://user-images.githubusercontent.com/18659734/86111962-4a856f00-babf-11ea-80d9-d98fe71ae94a.jpeg)

Starting to think my IDC connector may be dodgy
unfortunately now Ive broken a leg off the IC socket, pulling it in and out too many times. So will be back when I've got that replaced ....

lfitzsimmons, 2020-07-09:
Hey me again - just to let you know turns out my FPGA had bad soldering all down the GPIO header. Continuity seemed fine but must have looked ok because I was applying pressure to the pins. While the machine was running I pressed down on the pins and saw the picture improve in places so reflowed that side. 
 Looking good to me ! Fantastic picture in person through the OSSC. 
[image]

Thanks once again for all your hard work and for replying! Will test a lot and try to provide some feedback.

c0pperdragon, 2020-07-30:
Finally I got myself an RGB CRT monitor. Using the default RGBns palette from Hojo-Norem's software and it looks great now. This monitor accepts the compositve video signal from the A/V port as sync source, so no additional hardware is needed. 
Besides a properly soldered connector cable...
[image]

lyzanxia, 2020-08-19:
Just bought a mod thru videogameperfection. Installed it and have a nice picture, but am picking up noise from somewhere. I see a rolling down banner of noise across the entire screen. I have since disconnected gnd1 (only gnd3+4+5 are connected to gnd), but this did not help. Picture thru the din connector does not experience this rolling noise. Any idea what I could try to fix the noise?

c0pperdragon, 2020-08-19:
On such a "long" board, you should connect GND1, GND2, GND3, GND4  directly with pins down to the main board, so everything is firmly grounded.

When the noise is rolling through the screen this means the noise must come from the power supply (either internal or external), as it is obviously happening at rougly 50Hz. 
I would even guess it must be the 12volt rail that is generated internally by the C64 itself from the 9V AC line.

When you have a multimeter try to probe the various pins of REG1 on the mod board to see what voltage is comming in and what the regulator then makes of if. The incomming voltage should be a reasonably stable 9 to 12 volt, while the outgoing voltage should be a rock-soldid 3.3V.

lyzanxia, 2020-08-19:
I measured REG1. Input is 10,27 volt and output is 3.297 volt

c0pperdragon, 2020-08-19:
These voltages sound reasonable. Still, the noise must be comming from somewhere. Do you also happen to have an oscilloscope to again probe the 3.3 volt rail and see what is going on?

Another possibility is the component video circuitry in the TV itself. Do you have any other source for a component video signal to feed into the TV to check if this is working without noise?

lyzanxia, 2020-08-19:
Unfortunately do not own an oscilloscope. 
I am feeding the signal direct into a Dell monitor which has component input . Also tried thru a retrotink 2xpro to hdmi with the same results on another monitor. 

I follow your idea of ripple from the psu internal/external..but no idea how to proceed..

c0pperdragon, 2020-08-19:
Without an oscilloscope to hunt down the source of the noise, there is not much you can do, I guess. 
Maybe borrow a different type PSU (but do not use an original!) from someone or order it from amazon and send it back the next day...

lyzanxia, 2020-08-19:
Thank anyway for the help. I just rebuilt everything on a spare c64 board 250407 rev B. Exactly the same on this other board. So it has to be the psu which has a too high ripple I guess...

lyzanxia, 2020-08-24:
Received the new other “brand” powersupply and picture is now without interference. Crappy aftermarket powersupply I guess..

closed 2020-08-24 by c0pperdragon