c0pperdragon/C64-Video-Enhancement #28: Analog only board refdes error

TMS2764, 2020-04-19:
The analog board reference designators C10 and C29 are swapped on the PCB. According the the schematics, C10 is for composite pin 2 and C29 is for Luma pin 3. The video enhancement board PCB is ok.

[image]

c0pperdragon, 2020-04-20:
Good spot!
I have fixed this in the PCB layout and the gerber files as well.

c0pperdragon, 2020-04-20:
As this is the first time anyone seems to have actually built this board, it would be very nice if you could tell me how well this works for you (photographs are highly welcome also).

TMS2764, 2020-04-21:
I think its great. Glad I found this little analog version design of yours. Just what I wanted to replace those rusty RF modulators that just mess up the video signals.

[image]

I usually use it in S-video mode. Its difficult to capture but the image is super sharp. The moire effects are from the camera capture. There is no such thing when looking at the screen. There are some visible jail bars.

[image]

Here is a closeup of the text.

[image]

Here is the same text in composite video. Its not quite as sharp. Furthermore, there are some video artifacts that are visible but difficult to capture on camera.

[image]

I hope this helps. Thanks for the good work.

closed 2020-04-21 by TMS2764