c0pperdragon/C64-Video-Enhancement #29: Connector GPIO1 and RFCON1 overlap

TMS2764, 2020-04-19:
I am using a shrouded connector for GPIO1 and it is overlapping with the RFCON1 location. I know its not what was intended but I prefer a shrouded connector. It prevents me from plugging the cable backwards. I would recommend moving the connector a little bit toward the edge of the PCB next to the back of the case.

[image]

c0pperdragon, 2020-04-20:
As this is a very special requirement, I do not actually want to introduce a new PCB revision. Even more so, as the only pin that causes a problem is not even connected to anything on the FPGA board. 
I propose to just clip away this 4th pin.

TMS2764, 2020-04-20:
Fair enough. Keep it in mind for the next revision.

closed 2020-04-20 by TMS2764