c0pperdragon/C64-Video-Enhancement #31: Regulator REG1 gets pretty hot

TMS2764, 2020-05-01:
The FPGA regulator REG1 gets pretty hot. I would recommend connecting its TAB pin to a big copper shape to help it dissipate. Using a regulator for which the TAB is ground makes this simple.

c0pperdragon, 2020-05-01:
Yes, the regulator get a bit hot, when using the mod on a "long board" that provides 12V as its power input. But it is not dangerously hot - I can actually touch it without burning my finger. This is way inside the spec for this regulator.
Using a different regulator with better heat management would be an option, but then it needs to be mounted on the bottom side of the board, which makes assembly more difficult, especially when using a pick-and-place machine).

TMS2764, 2020-05-03:
On my C64 REG1 operates at about 56°C in a 22°C environment for a 34°C temperature rise. So you are right, technically nothing to worry about. But still, it’s the hottest component in the C64. I was simply suggesting of taking advantage of the large copper ground plane on the bottom to help dissipate some of the heat and bring the temperature down a bit. You don’t have to mount it on the bottom. By using a regulator that has ground on its TAB pin you can use a copper shape on top and transfer heat to the bottom shape by using several vias. Just an idea.

Top view
[image]

Bottom view
[image]

c0pperdragon, 2020-05-03:
I don't want to switch to different components in this phase of the project. This could raise just too much confusion. 
What I can do is change the board design to have a larger copper area on the top side for the 3.3V rail. Dissipating heat from the top is probably better anyway. 

While I am on it, I can as well relocate the GPIO1 connector a bit to allow usage of a shrouded variant.

c0pperdragon, 2020-05-15:
I have modified the board to have a large copper area for REG1 to work as a heat sink, and I have moved the GPIO1 connector a bit, so you can also use a shrouded socket.
This is now revision 5 of the board.

TMS2764, 2020-05-16:
Excellent. I have sent the files for fabrication. I will let you know how it turns out.

Zybex17, 2020-05-21:
Hi all, sorry if i disturb : is there anyone kind enough to provide one of this fantastic board to me? Please, let me know.

c0pperdragon, 2020-05-21:
Have a look at https://videogameperfection.com/2019/11/03/component-video-coming/
to get info about purchase options in the near future.

Zybex17, 2020-05-21:
Hey, hi, thank you very much for the quick reply: i appreciate alot!  I'll go to see then. Congratulation for the great piece of hardware: it's just awesome. It's many time that I'm waiting for exactly the same thing. Great job, really. :)

Inviato da Outlook Mobile<https://aka.ms/blhgte>

closed 2020-07-21 by c0pperdragon