c0pperdragon/C64-Video-Enhancement #32: NTSC and PAL interlaced mode

TMS2764, 2020-05-01:
Instead of having a scan line emulation mode or the 240p/288p mode I would recommend implementing an true interlaced mode such as 480i/576i 60Hz/50Hz. Vintage and many modern NTSC / PAL monitors only support interlaced mode specially when using component video.

c0pperdragon, 2020-05-01:
Interlace mode would not make much sense, as the C64 is inherently working in progressive mode.
Vintage displays (true analog CRTs) are normally perfectly fine with 240p/288p. 
Modern TVs have some trouble with this, but on the other hand they normally can show 480p/575p when comming through YPbPr.  Indeed this is actually a very well specified standard (in contrast to 240p/288p which is more of a hack).

TMS2764, 2020-05-03:
Look at it from the point of view of the display. I tried all three modes on my super-duper Dell 2707WFP and yes everything works well because it has a whiz-bang scaler built in. If I do the same on my Sony NTSC/PAL monitors PVM-14L2 (CRT) and LMD-1420 (LCD) then the 480p does not work at all. But you are right, the 240p does display a correct image. Concerning the fundamentally progressive image of the C64. Isn't that what the VIC-II is actually doing? Generate a 480 progressive image and displaying it interlaced? I simply believe that having an interlaced mode might offer greater compatibility with a variety of displays. Anyway, just my two cents worth.  Thanks again for sharing the project.

Hojo-Norem, 2020-05-03:
>  Concerning the fundamentally progressive image of the C64. Isn't that what the VIC-II is actually doing? Generate a 480 progressive image and displaying it interlaced?

No.  The VIC-II (and pretty much any other machine that outputs true 240/288p) do not generate the needed half line that switches between each of the interlaced fields.  On CRTs that leads to the electron gun passing over the same lines on each field, creating a progressive image.  I think Nintendo may have referred to this as 'double strike' output.

Thing is, I'm no expert but I guess you'd need a full frame buffer to do a proper interlaced output and the timing cost of that extra half-line will put the FPGA out of sync with the VIC-II and at the very least cause a regular stutter to the output.

closed 2020-05-15 by c0pperdragon