c0pperdragon/C64-Video-Enhancement #33: "BeamRacer" compatibility.

Hojo-Norem, 2020-06-02:
I just discovered a new upcoming VIC-II accessory by the name of 'BeamRacer'.
https://beamracer.net/site/

It appears to be a co-processor / memory module that sits in-between the VIC-II and the rest of the C64 system. 

My guess is that the video enchantment should still work with this as long as it sits between it the VIC-II.

What's your take on the matter?

c0pperdragon, 2020-06-02:
I guess it could basically work if stacked in the correct order: BeamRacer in main board, video enhancement on top of that and the VIC-II on top of that again (add a heat sink and this will really be quite a tower). 
But only if the BeamRacer uses similar signal timings as the CPU.  It is well possible that the VIC-II itself can accept more things than the CPU could ever generate. For example setting two registers in a single clock or other crazy things.  The video enhancement is not programmed to cope with this.

I guess  the only way to know is to actually try. And even then there may be very rare incompatibilities you find only after years of experimenting.

laubzega, 2020-06-05:
My intuition is that it should work. Since we have to support multiple VIC-II models, the timing of most accesses is rather conservative. One obvious exception is gray dot on/off switch, but I don't think it is important for users of this mod. :)

c0pperdragon, 2020-06-05:
I guess there is nothing anyone can do but try.
One thing will very probably not work: Changing the palette settings of the C64 Video Enhancement. I very much guess that the BeamRacer will not forward these custom register accesses. So to re-program the palette you would have to temporarily remove the BeamRacer.

c0pperdragon, 2020-06-05:
On the other hand: If changing palette is such a nice feature, maybe the BeamRacer developers can be persuaded to provide this kind of register forwarding....

laubzega, 2020-06-14:
After experimenting with a board kindly donated by c0pperdragon  I am pleased to announce that everything works correctly. Register forwarding is also operational, as long as BeamRacer is inactive (e.g. right after Run/Stop-Restore). I hope to work with @Hojo-Norem to add necessary deactivation/activation sequence to his palette editor, so that the process is fully transparent to the users.

Hojo-Norem, 2020-06-15:
> After experimenting with a board kindly donated by c0pperdragon I am pleased to announce that everything works correctly. Register forwarding is also operational, as long as BeamRacer is inactive [...]

That's great!  I'm guessing that what you are meaning by activation/deactivation is that I would have to detect if BeamRacer functions are activated first and deactivate them before continuing to unlock the video mod's registers?

laubzega, 2020-06-15:
Yeah, that's ideal:
1. Check if it's there.
2. Deactivate if present.
3. Play with palettes
4. Restore the original state.

But at the very minimum:

1. Deactivate (POKE $D02E, $40)
2. Play with palettes.

I will be soon updating our documentation on the subject (https://docs.beamracer.net/doku.php?id=introduction_to_programming_the_beamracer#initialization) with new register addresses (we decided to move them to avoid conflicts with software that pokes $D030 blindly).

laubzega, 2020-08-18:
Hello @Hojo-Norem. I am not sure if you have time to look into this, but just in case you do, the document mentioned in the previous comment has been updated with up-to-date info and register addresses. I do not anticipate further changes before the release.

Hojo-Norem, 2020-08-18:
@laubzega , I'll take a good look one I get the chance.  In the meantime, Firefox and Edge are refusing me entry. Edge reports 'DLG_FLAGS_SEC_CERT_DATE_INVALID' and Firefox reports 'SEC_ERROR_EXPIRED_CERTIFICATE'.

laubzega, 2020-08-18:
It'll be back up in an hour or so. M.

c0pperdragon, 2020-08-27:
I guess this issue is finished with the BeamReacer working as intended.

bodgit, 2020-10-11:
@laubzega I am planning to build the updated version of the c0pperdragon RF replacement board with the FPC/FFC connector so I can attach my BeamRacer directly. Could you tell me what sort of FFC cable you used? I'm guessing it's 0.5mm pitch with 20 contacts but you can get the cables with both sets of contacts on the same side, so it's pin 1 to pin 20, pin 2 to pin 19, etc. or on opposing sides which means pin 1 to pin 1, pin 2 to pin 2, etc. which is correct?

laubzega, 2020-10-11:
It is indeed 20-pin 0.5mm FPC, with contacts on the same side. Please see attached photo and top view of the layout for reference.

[image]
[image]

bodgit, 2020-10-11:
Excellent, thanks for the reply!