c0pperdragon/C64-Video-Enhancement #34: Question about video mod

guadax, 2020-06-05:
Hi,

just one question about your mod, sold by:
https://videogameperfection.com/products/commodore-component-video/

cannot understand this part of the description:

If 100% authenticity is important, for example for speed running or other competitions, the component video mod also includes the S-Video bypass (available separately here). This means you can use the best quality, cleanest S-Video output from the C64 at the same time as the component video mod. Use the S-Video out to game on your CRT while you stream or capture the YPbPr, great for speed runners! 

From this description seems that if you wants both composite and component video you have to buy another addon card:
https://videogameperfection.com/products/c64-svideo-bypass

I think that's not true ... or am I wrong?

Could you confirm this?

thanks

c0pperdragon, 2020-06-06:
The component Video Mod already contains everything. This other board is just a low-cost option for people wanting only the s-video improvement but not the component Video.

guadax, 2020-06-06:
Thank you c0pperdragon!

closed 2020-06-06 by guadax