c0pperdragon/C64-Video-Enhancement #36: Horizontal glitches with VIC-II 6569R3

tyristori, 2020-07-16:
I get horizontal glitches with 6569R3. Board is NO. 251137 REV C. Here's a video of the glitch:

[https://drive.google.com/file/d/1gpyFiY_BxD6BCISTT9gS2czUl1HwVugZ/view](url)

Problem is solved by installing 47pF capacitor between VIC pins 17 and 20.

c0pperdragon, 2020-07-16:
Another user had the same problem with the exact same chip revision. The solution with the 47pF capacitor really seems to be the best cure for this. I keep this issue thread visible in case other users also run into this problem.

c0pperdragon, 2020-07-23:
To anyone with a similar problem:
Please note that this capacitor should only be used if the VIC really creates a jittery clock signal. First probe the clock signal that goes into the FPGA on pin 20 of the ribbon cable. If the signal is a clean 50% rectangle, the capacitor will probably do more harm to the system stability than help.

XLRVIII, 2020-07-31:
I get vertical glitching with 6567R5. Board is NO. 251137. 
Composite and S-Video look fine. Component looks like a garbled mess....
[image]
[image]

c0pperdragon, 2020-07-31:
From the garbled picture I can only conclude, that you are actually using a  6567R56A  VIC chip which is currently not supported by the firmware. 
This revision was the very first that was used in any C64 and had  a timing of 64 clocks per line instead of the 65 clocks of all later NTSC variant  (all PAL-B variants use 63 clocks). This missmatch leads to the mod miss-interpreting all the digital signals and it then produces only garbage as video output.

The reason why this is not yet supported is of couse that I have no actual VIC for testing this and also nobody who yet contacted me, has. So now this would be the golden opportunity to do something 
about it.

But for this I would need your help. I could try to come up with a new firmware version which you would need to flash to the mod board and report back to me how this works. In order to flash a new firmware you would need to have a special programmer device, called the USB-Blaster. This can be cheaply obtained, at least there are some very cheap china-clones which cost about $5, including delivery. 
If you want to go that route (and other users with a 6567R56A chip would profit from this as well), please just create a new issue thread named "6567R56A compatibility", where we can discuss all this in more detail.

c0pperdragon, 2020-08-27:
For your interest: 6567R56A is now supported by firmware 2.9.