c0pperdragon/C64-Video-Enhancement #39: Missing lines in sprites

c0pperdragon, 2020-08-11:
Some users still report glitches in the display of sprites. The reason is the very narrow time frame during which the VIC's address signals are visible on the bus. This information is used by the FPGA to know if a given sprite is active in one line.
The correct way to solve this is to also implement all the logic from the VIC that decides about this sprite DMA. This involves quite some additional registers and counters and logic and I don't know if it fits in the already pretty crammed FPGA. 
I will probably tackle the issue in the intermediate future.

c0pperdragon, 2020-08-16:
@IanSB
@andriodious

I have redesigned the sprite DMA detection method, so it is no longer relying on exact timing.
This now also makes the "early sprite DMA" patch unnecessary. 

You can find the newest experimental firmware as usual at:
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/firmware

IanSB, 2020-08-16:
I've tried the new version but I'm still seeing very occasional glitches, maybe one every few mins although they seem to be different glitches to the missing lines. I think it's the sprite line moving to the right but they are so fleeting it's difficult to be certain what's going on.

I've seen these with both the R1 and R5 VIC IIs and I'm using the pixel clock from the FPGA. 

Also I still have the bridge on the two pins, do I need to remove that or is it ignored with the new design?

c0pperdragon, 2020-08-17:
I have further improved the sprite DMA detection to even less likely miss any activity signal. But I am not sure this is the problem you just mentioned. Anyway the new firmware is on its usual location.
Any no, you do not need the bridge any longer, but it also does no harm as it is just ignored.

c0pperdragon, 2020-08-17:
Do you have a means to record a full video where these glitches could be visible?

IanSB, 2020-08-17:
>Do you have a means to record a full video where these glitches could be visible?

I tried the latest 2.9 beta2 and still get glitches but it does look it's just line missing lines. 
I managed to record them using a HDMI to USB interface and here are a couple of screen caps:
(Missing line in the white player's head)
[image]
This was the frame immediately following the above which doesn't have the missing line:
[image]

c0pperdragon, 2020-08-17:
I am nearly at my wits end. I have uploaded a new firmware which does the DMA detection slightly different, but I don't think this would really help much.

But your observation of the glitch happening somehow deterministic now gives me a totally different idea. Maybe it is no glitch of my mod board at all, but a glitch of the game itself! It is very hard to observe, but I watched this scene on the original A/V out and it looked as if there is indeed a damaged line there. Could you record the A/V output for comparison as well?

IanSB, 2020-08-17:
Yes that happens with the analog output as well so it's a VIC II gitch, last time I did a comparison there were so many glitches on the digital output that I probably didn't notice an extra glitch or two on the analog output as well.
Here it is on the analog output:
[image]

So I guess it looks like this new build will fix the timing issues on my C64.

I'll give the beta3 version a try as well and report back.

IanSB, 2020-08-18:
I've tried 2.9 Beta 3 and not found any glitches other than ones that were on the analog output as well.
I did this by recording the normal analog output on my PC (via a retrotink) and every time I noticed a glitch on the YUV output I checked the normal output recording and found it also had the glitch. (Tried with both R1 & R5 chips)

So Beta 3 looks good.

As a final experiment, I downgraded to 2.9 beta 1 which had earlier detection code and I think I had one case where I saw a glitch which wasn't on the normal output recording but it's difficult to confirm as I can only record one output at a time so can't replay both outputs side by side.

Do you want me to try Beta 4?

c0pperdragon, 2020-08-18:
Thank you for testing the sprite handle so thoroughly!  So we can consider this issue resolved.

Beta 4 adds experimental support for the 6567R56A variants of the VIC  (see https://github.com/c0pperdragon/C64-Video-Enhancement/issues/40). When firmware is finally finished and working on the actual chip and all is well, then it makes sense to re-test the firmware on all other VICs as well.  
Until then it is not necessary to test it on anything than the intended chip variant.

closed 2020-08-18 by c0pperdragon