c0pperdragon/C64-Video-Enhancement #40: Compatiblity with 6567R56A

c0pperdragon, 2020-08-18:
Hi @laubzega,

As previously discussed,  I have made a first experimental firmware version to support the 6567R56A chips.
These are quite rare and probably no demo directly supports them (different cycles per line and lines per frame). Also there is not much documentation or test programs available. 
I think it is good enough if "normal" screen modes and sprites work , without the fancy stuff demo-coders normally use to trick the VIC into all kind of strange behaviour.

You can get this latest firmware from https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/firmware

To actually use the 6567R56A mode, you need to have the pixel clock running at the NTSC frequency (using a normal NTSC computer) and additionally you must pull pin 5 of the JTAG header to ground. Pin 10 and pin 2 of this header already provide ground, so a short jumper wire should work. Obviously you can only attach this jumper when the JTAG programmer is not attached ;-)

laubzega, 2020-08-20:
With the JTAG pins shorted the base image shows up correctly. Also simple demos playing with colors posed no problems (top image is S-Video, bottom - component):

[image]

However, as I moved to more complex things, it became apparent that not everything is behaving as it should, as this FLI image demonstrates:

[image]

What you cannot see here is that there is some blinking in the image, suggesting there is some timing component to it. Here is the same  demo working correctly with 6567R8:

[image]

If you would like me to run some specific test software or demos, just send me a link.

laubzega, 2020-08-20:
One more thing - I have tested two 6767R56A VICs, with identical results.

c0pperdragon, 2020-08-20:
I am actually a bit surprised that it works so well already, as I had only a quity sketchy doc and guesswork.

Before diving deeper into thr FLI topic I would like to ser how sprites behave. Just try a few games and see if everything is actually visible and at the right places.

laubzega, 2020-08-23:
Yup, that was it. With the latest firmware I am no longer seeing issues with Bruce Lee and BoulderDash title screens. Also FLI demo looks correctly.

Ghosting is still there after palette flashing.

c0pperdragon, 2020-08-26:
Oh, my! The FPGA is so totally at its limit that even the slightest changes could cause the VDHL fitter to no longer be able to fit the logic into the device.
So I had to refactor quite some parts to squeeze out a few more LUTs to actually make the changes...

I have included the discussed changes and also fixed the ghosting, at least I hope so, because in my current setup I can not test this...
I also tried to vertically move the display a few lines, in the correct direction, I hope...

Please give the latest firmware (beta7) a try on the  6567R65A. And if this now all looks good, please re-test it on other VICS as well, as I could very well have broken quite some things there...

laubzega, 2020-08-27:
I have a love/hate relationship with logic squeezing. Sometimes it's very rewarding, sometimes it's a major PITA. At least you have a clear migration path to 10M04. Not so with the BeamRacer. :/

Anyway, beta7 works great. I tested both R8 and R56A, no issues observed. Colors align correctly after reprogramming, the vertical shift is no longer there. Games and simple demos I tried worked fine.

Thank you so much for all this work!

c0pperdragon, 2020-08-27:
Thank you very much for all this testing.
I have released 2.9 which should now indeed cover all possible variants of the VIC-II.

I am pretty sure I will not migrate to a larger FPGA. There have been too many boards produced for which I will not give up support. As it seems the feature set is now complete, anyway.

closed 2020-08-27 by c0pperdragon