c0pperdragon/C64-Video-Enhancement #41: Glitches/Interference with analog only board.

eflopez, 2020-08-19:
I put together a couple of the analog only boards but they both end up with glitchy video. On one, I get good video for about 20-30 seconds and then it starts to have these issues. The other board is no good pretty much straight away. They both exhibit the same artefacts though.

I don't have a lot experience with electronics, so wondering if I've done something wrong, and not sure how to investigate further. 

This is for a PAL 250469 (No. 252311 REV.4) shortboard with composite video on a CRT TV.

Any ideas?

[image]
[image]

c0pperdragon, 2020-08-19:
If you are referring to this diagonal stripe of light and the neighbouring vertical strip, I would really have not the slightest idea. But I guess this is just a reflection from your room, door  isn't it?

But I don't know where the rest of the noise comes from. The whole analog signal path of the VIC is pretty bad already, and composite video even more so. The analog video board can not really do much about this. But maybe there is something specificly wrong with your installation. Please make a picture of your install, so I could better guess what is going on.

eflopez, 2020-08-19:
Yep, diagonal line and vertical strip are just a reflection from the window :)

I've uploaded some photos and a video here:
https://drive.google.com/drive/folders/1gSFHNOk5FvQe_4_FEl2io-MgXK0Rk7pV

You can see the noise start at about 14 seconds, before that the output looks quite good.
Since it works well for a bit, my hunch is that as some component warms up, it starts to cause problems.

Please excuse my poor soldering, first time doing SMD. 

I did have to substitute the LM1117 which wasn't in stock locally with a TLV1117 which according to the LM1117 datasheet is s drop in replacement could this be the problem?

c0pperdragon, 2020-08-20:
The regulator you are using as very similar specs to my original one. Maybe a very slighly higher drop-out voltage, which could actualy worsen the effect of voltage ripple on the 5V supply line. With a 1.3 drop-out the margin between 5V and 3.3V is already nearly exhausted further degradation will probably show.

The horizontal distance of the vertical noise bands looks like it is caused by some supply voltage ripple of about 100kHz.  This hints at a switching power supply with insufficient ripple suppression. 
That the effect starts only after some time may be caused by the PSU heating up. 
Do you have an oscillator to probe the 5V line (rightmost pin on your board) to verify this? 

Looking at the schematics of your main board, there is the filter capacitor C58 that should already reduce incomming ripple. Maybe this has gone bad already or just is not up to the taks with your PSU. You could try to replace this with a newer and higher-capacitance piece. 470uF could probably improve the matter.

c0pperdragon, 2020-08-20:
oops - of course I meant "oscilloscope", not "oscillator"

eflopez, 2020-08-26:
Hey, I replaced the regulator with the LM1117 and things are working amazingly! Composite looks great too.

closed 2020-08-26 by eflopez