c0pperdragon/C64-Video-Enhancement #45: Default color plaette / How to flash a firmware

ei8htthepG, 2020-09-22:
Hi,

I've just installed the mod today and experimented with the color editor. Now the colors are more like the original but to be honest, I've liked the new fresh ones more. How can I get them back?

How Do I flash a new firmware on the board?

c0pperdragon, 2020-09-22:
The color editor has an option to revert to the default colors. If I remember correctly this is some option after the program start.

If you want to flash a newer firmware, you need to get a USB-Blaster device (realtively cheap from various chinese suppliers). But this is probably only necessary if you experience any visual problems. The version 2.6 is already compatible with all games and demos you can  throw at it. Later firmware revisions were necessary to address glitches that occured only with specific machines.

ei8htthepG, 2020-09-22:
Thx...I made it!

Your product is awsome!

ikrananka, 2020-11-27:
I have a cheap Chinese Xilinx JTAG programmer similar to this one [https://www.ebay.ca/itm/Xilinx-Platform-USB-Download-Cable-Jtag-Programmer-for-FPGA-CPLD-C-Mod-XC2C64A/391876366607?hash=item5b3da6d10f:g:lowAAOxyXzxTGNZ0].

Is this suitable for flashing the firmware?

c0pperdragon, 2020-11-27:
No, xilinx programmers can not be used for intel (former altera) FPGAs. You need an USB-Blaster.

ikrananka, 2020-11-27:
Thanks.  I'll get hold of a USB Blaster.

ikrananka, 2020-11-27:
This [one](https://www.amazon.com/RioRand-Blaster-CPLD-FPGA-programmer/dp/B00IRODADK) seems to be appropriate.  Appears to support the MAX 10 and gets good reviews.

c0pperdragon, 2020-11-27:
This looks very good.

sedorrr, 2021-04-24:
Which software do you use for programming? I cant get Quartera... is there an alternative?

c0pperdragon, 2021-04-24:
I use the Quartus Prime Lite edition which is a free download, available from: 
https://fpgasoftware.intel.com/20.1.1/?edition=lite&platform=windows

sedorrr, 2021-04-24:
Thanks, that's what I am trying to get - but seems registering is not working, will continue to get it :)

Hojo-Norem, 2021-04-24:
You might have some luck using one of the logins listed on Bugmenot.  I think that's what I did.

http://bugmenot.com/view/fpgasoftware.intel.com

sedorrr, 2021-04-25:
> You might have some luck using one of the logins listed on Bugmenot. I think that's what I did.
> 

Good idea... "seems" to work ;)

closed 2022-02-11 by c0pperdragon