c0pperdragon/C64-Video-Enhancement #46: Mod not working anymore

ei8htthepG, 2020-10-09:
Hi,

my mod isn't working anymore. After I turn the c64 on there is just a weird picture on my BVM and no signal on a second monitor for test cases. The c64 displays a picture via S-Video. The chip on the mod is getting very hot (can't touch it after 5 seconds) shortly after I started the c64. I am a bit afraid. Can I check voltage somewere?

c0pperdragon, 2020-10-09:
If the FPGA (the big square chip) gets hot, this is really a very bad sign. Normally it should not get more then mildly warm even after prolonged use.
You could actually measure the voltages that go into the FPGA on the voltage regulator REG1: Measure voltage between the big tab of this chip and some ground (cartridge slot metal for example). This should have a solid 3.3V.
If the voltage is too low already there is probably a short somewhere or maybe the chip has some internal fault and now burns itself to death. 
You could also try to unplug the ribbon cable that goes to the VIC in the case that the short happens there. 

In any case I have not much hope that this device can be repaired. 
If you got it from videogameperfection, you could ask for a replacement. I don't know about the exact terms but maybe this is covered by warranty...

ei8htthepG, 2020-10-10:
I did some more tests. The FPGA gets hot, when I plug in the ribbon cable. I will contact videogameperfection and hope for warranty. Thx for help.

c0pperdragon, 2020-10-10:
Hmm. Unplugging the ribbon cable actually did help? Maybe the fault is then indeed with the adapter board, so you should also change this to make sure.
If you have some oscilloscope you could actually check the voltages of the signals that come through the cable. Should all be 3.3V signals. 

For completeness, could you post a picture of your install?  I would be escpecially interested on the GND connections of the FPGA board - as this chould be a major failure point if not done properly.

closed 2021-01-09 by c0pperdragon