c0pperdragon/C64-Video-Enhancement #49: Wiggly screens rendering issue.

chabiss, 2020-10-15:
Finally installed my board from VGP inside 250407 REV C board using a 6569R5 VIC chip.  The results are wiggly. See the video below:

https://youtu.be/WhkywaEn0hI

and a shot picture

[image]

I've tried different monitors and they all have the same problem.  I've noticed also that some of the C64 demos (like Lunatico) produce some very weird artifacts. Since the board is on 2.6, I assume that I'll need to get to 2.9 to see any improvement of Sprites rendering timing? 

But that wiggly rendering on component output is weird.  Note that the 64 S-Video coming out of the board is fine.  Only the component output has this issue. 

Any idea what could cause this?  Note that this is a 250407 REV C NTSC board that has been converted into PAL by changing the crystal. The machine runs fine in PAL, but the AC 9 volts from the power supply will still come into the machine at 60 HZ.  I'm wondering if this has anything to do with the issue I'm seeing?

chabiss, 2020-10-15:
Hmmm... This could very well be the same issue as https://github.com/c0pperdragon/C64-Video-Enhancement/issues/36

laubzega, 2020-10-15:
Unlikely. The screen with #36 is completely unreadable and skewed to one side. Here is just minor wiggling left and right.

chabiss, 2020-10-16:
@laubzega  you need to watch the video at
https://drive.google.com/file/d/1gpyFiY_BxD6BCISTT9gS2czUl1HwVugZ/view
, not the picture.  The symptoms are very similar to mine.

laubzega, 2020-10-16:
Then indeed. Apparently #36 is about two distinct issues.

c0pperdragon, 2020-10-16:
Yes, the second problem posted into this issue was completely unrelated to the original post. I have now removed those entries to not confuse the matter any longer.

If you have tried the 47pF treatment and it did not improve, I still have a second idea. Your board revision seems to have a clock circuit made from individual chips instead of the later integrated solution. This is known to cause a lot of troubles. 
Somes of the problems were fixed with later firmware revisions - especially sprite rendering.

If nothing of this helps, you can also use a special feature of the FPGA board that produces a pixel clock you could feed into the VIC-II to bypass the shaky clock generator - this is explained in more detail in https://github.com/c0pperdragon/C64-Video-Enhancement/issues/38  - but it requires a bit of rework of the C64 main board.

chabiss, 2020-10-17:
Yep... I fully read https://github.com/c0pperdragon/C64-Video-Enhancement/issues/38 and I'm 99.9% sure I'm hit with the same sparkly pixels issue :) It's going to be a fun project to diagnose. But first I'll move the firmware to 2.9

chabiss, 2020-10-22:
I moved my firmware to 2.9 and did the pixel clock mod.   

[image]

The mod did the fix for me. No more sparkling pixel! I threw everything at it: Sprite Crunch, VSP. It's rock-solid!!! I'm blown away by the board!!!!