c0pperdragon/C64-Video-Enhancement #50: Graphics glitches in Edge of disgrace

sivu, 2020-10-28:
In [Edge of disgrace](https://csdb.dk/release/?id=72550) there are couple of places where some pixels that are not behaving as they should, usually in the top or bottom left of the effect.

example from around [this part](https://youtu.be/nLIUkBa_mA0?t=66)
[image]

I am running 2.9 firmware.

c0pperdragon, 2020-10-29:
I know that this Demo actually has some real glitches which are also visible on the original s-video/composite output. 
These often happen in the various border regions and are sometimes not visible on analog screens because they are too far in the overscan region or too faint to be propperly perceived. As far as I know, the mod board truthfully reproduces all of those glitches. If you have a way to capture/view the original analog video alongside the mod output, please try to make such a direct comparison.

sivu, 2020-10-29:
Yeah, I will try to capture the comparison. I think I also saw some glitchy pixels further away from the border, I will try to capture those too.

sivu, 2020-11-05:
Compared to svideo out and same issues are seen so closing this

closed 2020-11-05 by sivu