c0pperdragon/C64-Video-Enhancement #52: C64 326298 REV A Question

ksliman, 2020-11-11:
So I desoldered the RF modulator but the holes on board dont match up. 
[image] and this is the FPGA board I ordered 

[image]

The RF Modulator was connected to these holes 

[image]

Now before I go and mess this 40 year old computer up , I figured I would ask ? would these connections work ? 

[image]

As it stands, when I position the board over the main board the wholes dont line up , so I have to solder wires.

c0pperdragon, 2020-11-11:
Wow! You actually seem to have the very first board revision for the 64. I was not aware of the fact that this board has a different layout for the connectors to the RF modulator. So you obviously cannot use pins to directly connect it.  As I don't know the assignment of the various solder holes, you need to do a bit of probing with a multimeter to find out what is what.
I will give you more specific instructions later today.

ksliman, 2020-11-11:
Thank so much! I just happen to find this C64 in a goodwill store a while back and its been sitting my attack. I figure I would play with it a bit.  I did some probing on the board and got some measurements that may or may not help. 

First Here is the RF Modulator 

[image]

Also when the RF Modulator was connected the pins in are labled VIDEO, AUDIO, B+ 

[image]

All 3 of those pins measured 5 Volts DC 

also on the side of the square I see 4 pins that were not connected to the RF Modulator that measure 5 vdc and gnd 

[image]

c0pperdragon, 2020-11-11:
I had a look at a schematics diagram I found online, and it very much seems that this specific model makes is quite easy to install the mod. The analog signal path from the VIC to the A/V connector is not routed through the RF modulator at all (contrary to all other C64 models), so the A/V connector should be still functional even if the RF modulator is now removed. 
No need to bother wiring up any of these signals to the FPGA board. 
You only need to connect the 4 GND connections on the left and right edge of the FPGA board to the mouting holes of the former RF modulator. These may actual be in the correct place, or you need to become creative to make the connections match. This connections also serves as the mechanical mouting points.
Additionally you need to connect pin 1 of RFCON2 (the rightmost pin with the square pad) to a positive supply somwhere on the main board. Anything from +5 to +12V will do, but I recommend the +5V supply to not to create too much waste heat. Make sure this is really a supply line and does not just carry this voltage as an output signal.
Then just attach the ribbon cable from the VIC II adapter in the right orienation (red wire on topmost position).

c0pperdragon, 2020-11-11:
As you have this very early computer, chances are that it also has the very first variant of the VIC II chip. I can not read the version number on any of your pictures, but if it is revisions 656765A, you will have to take some additional steps to make it work.
The firmware 2.6 that is preinstalled on the boards from videogameperfection.com  does not yet support this chip and needs and update. I can instruct you on how to do that if necessary.

ksliman, 2020-11-11:
Man that sound great! ok here is the the VIC chip if I am reading that right its 6567R56A

[image]

c0pperdragon, 2020-11-11:
So in order to use the mod together with this VIC revision, you need an updated firmware. Unluckily this requires a dedicated programming device (a so-called "USB-Blaster"). The original device from Intel is very expensive, but perfectly working clones can be had from China really cheaply (like for example https://www.ebay.de/itm/322658807923).

But before this you could first try to use the firmware as it is and take a picture of the screen. If my guess is correct it should look totally mangled in a very distinct way.

ksliman, 2020-11-11:
Sounds good I will order the device.. I will post a picture here of the mangled screen. Thanks for all you help.

ksliman, 2020-11-11:
Will this work! https://www.amazon.com/Blaster-Download-Programmer-Debugger-XYGStudy/dp/B00CAVG4JS/ref=sr_1_10?dchild=1&keywords=Altera+Mini+Usb+Blaster+Cable+For+CPLD+FPGA+NIOS+JTAG+Altera+Programmer&qid=1605131499&refinements=p_85%3A2470955011&rnid=2470954011&rps=1&sr=8-10

I trust amazon since  and they will have this for me tomorrow

c0pperdragon, 2020-11-12:
Guessing from the compatibility list, this may be be OK, even if they don't state your specific FPGA explicitely.
In order to use the programmer you also need to install the "Quartus Prime" software suite which is available for windows and linux. Don't know if you happen to only have a Mac - if so, you would need to find some compatible computer to do the firmware upgrade.

ksliman, 2020-11-12:
Thanks , ok I will find the software! I also had to order a longer 20 pin cable since the placement of the VIC chip is different on this particular board, the supplied cable was few centimeters too short.

c0pperdragon, 2020-11-12:
About the cable: You could gain a centimeter or so by cutting away the strain releave on the connector. Maybe this is already enough...
You really have the oldest and rarest C64 board variant her and it is a real miracle that it still works.  It actually does, doesn't it?

ksliman, 2020-11-12:
Yeah, it works. The only thing that was bad was video quality was really bad! which lead me to find this device! the only other thing I have to do the C64 is clean up the keyboard as some of keys need to be pressed really hard to function.

tyristori, 2020-11-14:
Here's an explanation and fix from Ray Carlsen for poor video quality of 326298:

>WEAK VIDEO OUT OF THE EARLY C64 326298 MOTHERBOARD
Latest updates or corrections 7-13-2018

>I always wondered why this particular motherboard put out such a weak 
video signal. I have four of them and they all do it. The early VIC chip 
itself has a characteristic flaw, vertical "ghost" lines that closely 
follow each screen character. That IC is the ceramic version 6567R56A and 
they all do it. The later CBM black epoxy case 6567 IC outputs clean video. 
However, the weak video persisted even after the VIC chip was replaced. 
     I recently had some free time so I decided to find out why those early 
boards put out video that is faint and blurry. It turned out to be a 
factory mistake. There are several components between the output line of 
the VIC chip pin 15 and the two video sources (luminance and composite) 
that feed the A/V jack. There is a buffer transistor Q4 wired as an emitter 
follower, a low impedance source to drive the outputs. Measured with an 
oscilloscope, the luminance signal looked OK but the level of the composite 
output was less than half normal. It should be at least 1VPP at the jack 
when connected to a monitor. Looking at the schematics (I have several from 
different sources), they all show the value of resistor R10 to be 120 ohms, 
but in my four boards that resistor is 300 ohms! No wonder the video is so 
weak. There are two other resistors that are shown on the schematics but 
that don't exist on the boards, namely R48 and R49. The signal on coil L3 
goes directly to pin 3 (no series resistor) of the video modulator and there 
is no resistor to ground there either. 
     So, how do we fix the weak video problem? Inside the video cage, just 
to the right of the VIC chip, I connected a 220 ohm resistor across R10 to 
bring the video level up where it belongs. Easy and quick. (See photo) This 
old board now looks as good as the later version C64's. 

>Ray

[image]

ksliman, 2020-11-15:
I will add a resistor to it as well. I had to reorder the FPGA cause I screws up the solder pads when trying to de-solder the attached cable. The new one should be here on Tuesday . I will  add the resistor now since I already have it open.

tyristori, 2020-11-15:
My board is from VGP too and I made a longer cable. I also added a right angle IDC-header as I don't like permanently soldered ribbon cables.

[image]

ksliman, 2020-11-15:
Oh I agree , they should have not permanently attached that cable. Where did you get that heat sink from ?

tyristori, 2020-11-15:
From Retroleum. They sell heatsink kits for C64.

ksliman, 2020-11-20:
So I finally got it all hooked up, and I updated to the latest firmware. But when I turn it on all I get is this 

[image]

Any Ideas?

c0pperdragon, 2020-11-20:
Yes, this is the video output when the mod interpretes the signals of the 6567R56A is if it were a standard NTSC chip variant. Because the mod (even with the latest firmware) can not detect by itself that this chip is used, you need to give it a little hint. For this, plese refer to the section "Compatibility" on the documentation page:  https://github.com/c0pperdragon/C64-Video-Enhancement. I think I have a small typo there (65A instead of 56A), but it is this very chip I am refering to.

ksliman, 2020-11-20:
I tried it that did not make any difference.  But just to be sure I jumped these two pins 

[image]

Is this correct?

c0pperdragon, 2020-11-20:
This is not correct. 
For two-row connectors the pin ordering is always going in zig-zag between both rows. So in your picture pin 1 is bottom right, pin 2 is bottom left (which you correctly used to get GND voltage), pin 3 is directly above pin 1 and the relevant pin 5 is above pin 3 (in the middle of the right column).  
You did connect pin 2 to pin 10, both of which are GND so you were lucky not do damage anything.,
Connect pin 2 to pin 5 instead.

ksliman, 2020-11-20:
Man I am glad I have two of these things! but ok before I mess this up I would jump these 

[image]

c0pperdragon, 2020-11-20:
Yes, this is correct.

ksliman, 2020-11-20:
Wow , this looks so good ... Man thank you so much for all the help. 

[image]

closed 2020-11-21 by ksliman

woelfto, 2022-03-23:
this is incredibly helpful, I have the same board and was bummed last night when I realized the pins didn't line up.  Qq, did you end up having to install the firmware upgrade or did just jumping the pins fix it?