c0pperdragon/C64-Video-Enhancement #55: artifacts and how to program

a7mag3ddon, 2021-02-10:
HELLO

I have a pal c64 with longboard and im getring artifacts on the screen for a few minutes after turning on and sometimes im missing red from the screen.

On another forum (Video Game Perfection) someone else was gerring same artifacts aand fixed this with a firmware upgrade so was wondering if theres any instructions for how to flash my board ?

The second question i have, is there a way to see what version of FW is on my board? It was purchased around middle of NOV and fitted around the 4th Dec from VGP.

Thanks

c0pperdragon, 2021-02-10:
There is no direct way to know which firmware is flashed to the board. But if you have it from videogameperfection.com it is surely 2.6. 

To upgrade the firmware you need to purchase a small programmer device, the "USB-Blaster", which can be bought fairly cheap. 
You also need the "Quartus Prime Lite" software to do the programming from the PC  (free download but quite big).

Which long board revision do you have? The very first ones (with the non-integrated clocking circuit) are known to cause problems like this. The newest firmware works around most of these issues.

a7mag3ddon, 2021-02-11:
I have a usb blaster arriving today and i have the intel fpga software and max10 addon at the ready, my board i thin ends 407 i will dig out some photos of internals and take a look as im not with machine at the moment.

a7mag3ddon, 2021-02-11:
apologies for the delay, the assy no is 250401  art : 251137 Rev C which i assume is a very common PAL machine

Going to flash/program the fpga in 10 mins.. fingers crossed.

a7mag3ddon, 2021-02-11:
Ok all flashed, and put a small sticker on saying FW 2.10 with the date. Hope thats sorted my 2 problems out, lack of RED and artifacting.

c0pperdragon, 2021-02-11:
Guessing from this list of board revisions

http://www.cbmhardware.de/show.php?r=1&id=25/Die%20Platinen%20des%20C64

this seems to be one of the early boards with the multi-chip clock circuit. 
This is known to cause problems with the 2.6 firmware. If 2.10 does not already fix this, there would be an additional tweak to use a cleaner clock. As added bonus this often also cures the infamous VSP-bug.

a7mag3ddon, 2021-02-11:
Where is the info on the cleaner clock mod? I will take a look and see if its worth doing.

a7mag3ddon, 2021-02-11:
I think this is what i need ?

https://www.ebay.co.uk/itm/MOS8701-Clock-Chip-Replacement-Commodore-C64-C128-small-size-big-performance-/133504255823?_trksid=p2349624.m46890.l49292

ikrananka, 2021-02-11:
Isn't your Assy No.250407 and NOT 250401 as you state above?

The 250407 uses an older, more complex, clock circuit compared to later motherboard versions.  The eBay auction you link to is for a replacement of the single clock chip in motherboards later than the 250407, i.e. that part is not compatible with your motherboard.

a7mag3ddon, 2021-02-11:
Damn, what options do i have?

ikrananka, 2021-02-11:
If after upgrading to the latest firmware you're still getting graphical issues then you'll likely need to do the "pixel clock mod".  Check out Issue 53 for this GitHub project where I did it on my 250407 board.

The original post detailing the procedure is here: https://github.com/c0pperdragon/C64-Video-Enhancement/issues/38#issuecomment-663986704

a7mag3ddon, 2021-02-11:
Its all getting a bit complicated now, i need to solder a wire from R24 and connect it to pin 6 of the lifted 8701 ?

ikrananka, 2021-02-11:
There is no 8701 on your revision of the motherboard so that is incorrect.

On the 250407 board you, in effect, need to disconnect pin 7 of the 74LS629N (U31) and connect your new wire to the motherboard where that pin was previously connected to.  If you want to avoid any desoldering then the simplest route is to cut the right-hand leg of FB17 and lift up FB17 to move it out of the way (Note that FB17 is connected to pin 7 of U31 hence why this is an alternative solution).  Then solder a wire from the left side of R24 (see the photos to make sure) to where the right-hand side of FB17 went into the motherboard.  Check out the third photo in this comment : https://github.com/c0pperdragon/C64-Video-Enhancement/issues/38#issuecomment-671067022 to verify the correct locations.

Note that it is recommended that you include a 100ohm resistor in this new connection which you'll see in the photos.

For me, I preferred to retain the FB17 in place and still have it active in case it might still help eliminate a small amount of noise from the signal.  So, I desoldered U31 completely and soldered my new wire to the left-hand side of FB17.

a7mag3ddon, 2021-02-11:
Oh yeah, aorry i confused myself there.

I will try this mod and see if it fixes my lines problem when cold i can cleay see FB17 in thay photo in the other thread.

Do you know what R27 does on this board ,? Do i need it ?

ikrananka, 2021-02-11:
> Do you know what R27 does on this board ,? Do i need it ?

That's something @c0pperdragon will have to answer.

c0pperdragon, 2021-02-12:
This trimmer seems to be used to fine-tune the oscillator. This is necessary to get the exact PAL/NTSC subcarrier frequency. The pixel clock is then derived from this.
When you take the pixel clock from the FPGA, the subcarrier frequency is still used to produce the color signal for the analog A/V port.

closed 2021-04-23 by c0pperdragon