c0pperdragon/C64-Video-Enhancement #56: RGBns with Fix for Glitches (PAL)

ottj, 2021-02-22:
Hi
I have a PAL 250407 with a VICII 6569R1. Using the stock firmware of the mod I encountered the usual errors and checked the fixes here.
I ordered a USB blaster to update the firmware and plan to connect R25 on the mod board via 100R to the pad of the desoldered right leg of FB17. 

What is unclear to me though: 
1. should I also replace the crystal with a new one?
2. (most important): I plan to use this C64 in RGBns mode, hence take the sync from the AV port. Is this still possible after performing this fix?

Thanks in advance

c0pperdragon, 2021-02-22:
No need to do anything with the crystal. When you do the mentioned changes, the VIC II and the whole system will be clocked from the FPGA board. The crystal is then only used to create the color signal on the composite/s-video output.

The signals on the A/V port are always in prefect synchronization with the VIC II and the output of the FPGA board. It is absolutely fine to use either the luma or composite signal for  sync and the RGB lines for color. Depending on your monitor you may not even use the no-sync mode and just use the RGsB for this. My monitor at least can perfectly ignore the sync on the G line.

ottj, 2021-02-22:
thanks for the fast response! Ok, that is good to hear. I will proceed with the normal fix then, as soon as all the part have arrived.
I require proper RGBS for my Scart switcher (24 port Hydra), hence I cannot use YPbPr or RGsB, and I want to avoid an additional transcoding step (e.g. Comp2RGB).

ottj, 2021-02-28:
I was able to flash the newest firmware (2.10, Quartus v20. drivers did not work, I had to use version 18 to be able to install the drivers for the USB blaster). With this firmware the glitches were gone but the blue colors were completely missing from the picture. I then proceeded to do the sync fix using a 180 Ohm resistor since I did not have a 100 and you mentioned the value does not matter that much. I now have a perfect picture, also in RGBns mode! 
Thanks for creating such an awesome mod and the fantastic support!
[image]

c0pperdragon, 2021-02-28:
My congratulations.  This looks very good.
The ability to use these high-quality analog screens was a part of the motivation to go for analog output instead of HDMI.

tyristori, 2021-02-28:
How do you actually enable the mod to output RGB instead of component video?

ottj, 2021-02-28:
@tyristori You can use the [palette editor](https://github.com/Hojo-Norem/c0pperdragon-VIC-II-Palette-editor) to set your preferred output mode (YPbPr a.k.a component, RGsB a.k.a. sync-over-green, or RGBns a.k.a normal RGB signals and a separate sync from the AV port. I built a cable that has the Scart connector on one side and on the other 3 RCA plugs to plug into the output of the mod board and a DIN 8 U-shape plug to get composite video as sync signal and sound from the AV port of the C64.

tyristori, 2021-02-28:
Thanks ottj.

Icelvlan88, 2021-02-28:
Can you post a picture of to cable? I’m using component with a mini din cable but have been thinking of switching it. What components din you put inside cable?

ottj, 2021-02-28:
@Icelvlan88 Here is a picture of the cable. I connected the color RCAs with their respective RGB pin and ground, e.g. RCA red to scart_red, red_ground to scart_red_ground etc. The U-shaped DIN8 pin is connected like this: DIN pin 4 (composite video) to Scart pin 20 (composite video in). Din pin 3 (audio out) to Scart pin 2 and 6 (audio in L + R). Din pin 2 (ground) to Scart pin 17 and 4 (video ground and audio ground). I did not add any other components (resistors or capacitors).
[image]

Icelvlan88, 2021-02-28:
What are you doing for your sync?

ottj, 2021-03-01:
@Icelvlan88 I use the composite video output on the AV port (Pin 4) as sync signal. You could also use Luma. If your equipment requires a clean sync signal you may add a sync cleaner to your Scart plug. Of course, then you also need to add an additional wire (5V) from the AV port to the Scart plug to power the cleaner circuit.
You find such Scart sync cleaners here:

[https://videogameperfection.com/products/sync-in-scart-board/](https://videogameperfection.com/products/sync-in-scart-board/)
or here:
[https://www.retrogamingcables.co.uk/SYNC-SEPARATOR-CIRCUIT-BOARD-LM1881-COMPOSITE-SYNC-CSYNC-PCB-STRIPPER?search=sync](https://www.retrogamingcables.co.uk/SYNC-SEPARATOR-CIRCUIT-BOARD-LM1881-COMPOSITE-SYNC-CSYNC-PCB-STRIPPER?search=sync)

or here:
[https://console5.com/store/lm1881-pcb-bundle.html](https://console5.com/store/lm1881-pcb-bundle.html)

hugo-lyppens, 2022-03-20:
I would like to use RGBns to connect my c0pperdragon C64 to a Dell 2001FP 20" monitor. I doubt this monitor accepts SyncOnGreen. This monitor is especially suited to retro computing, because it syncs down to 15kHz, supporting all c0pperdragon mod modes. I would need TTL level sync signals. Are Hsync and Vsync available as outputs on the FPGA? If so, which pins?

at this link TTL sync is listed as one of the options.
[https://videogameperfection.com/forums/topic/rgb-options-for-the-c64-component-video-mod/](url)

but it is not clear to me where the sync signals can be found. Thanks.

c0pperdragon, 2022-03-20:
@hugo-lyppens 
The mod board does not provide seperate hsync and vsync. Everything is done here with csync.

If you want to attach a monitor that absolutely needs hsync/vsync you may do that with a bit of additional hardware. Maybe the LM1881 sync stripper is the right choice here? In this case reprogramm the palette to RGBns and use the LM1881 to create the necessary sync signals from the original A/V output.

hugo-lyppens, 2022-03-20:
Thanks for responding. I would like my monitor to work regardless whether I use 15kHz line frequency or doubled line frequency since it supports all such a wide range of line frequencies. The original A/V output is always 15 kHz, even if the mod is line doubling, right? So that won't work. It sounds like I would have to follow the instructions here:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/58#issuecomment-795722678

and then use the sync stripper on top of that. The issue with that is that I would have to add a switch to the back of my C64, because I would want to keep the current setup when connecting to Component displays and the new setup when connecting to VGA with separate TTL sync.

Aren't there any spare outputs on the fpga to output hsync and vsync? Wouldn't that be a lot easier (assuming these signals are present inside the fpga as part of all the logic).

c0pperdragon, 2022-03-20:
Well, there are some spare pins on the FPGA, but the firmware does not create hsync and vsync internally. Also I have used up the internal FPGA resource to about 100% after countless feature requests already, so there is not much room to do changes anymore.

If your monitor can actually ignore the sync tip on an RGsB signal (some do use the idle level just after the sync tip as their black reference), you can use this as its RGB input. The mentioned sync seperator can extract csync/vsync from the RGsB just well. You can build this into the C64 or create an adapter (need to be powered).

hugo-lyppens, 2022-03-21:
I just made a welcome discovery: the Dell 2001FP I have supports sync-on-green. So I just made a cable with only RGB pins connected and it displays the C64 screen! Now I just have to switch from YPbPr to RGB using your palette prg. Thanks again.