c0pperdragon/C64-Video-Enhancement #57: hdmi output like for Amiga

mistepien, 2021-03-06:
Is it possible to extend the project to hdmi port using rpi0?

c0pperdragon, 2021-03-07:
Actually the RGBtoHDMI upscaler already supports the C64. This is an external device that received the analog signals from the component video mod and creates as pixel-perfect HDMI signal from it in the same quality as for the Amiga now.
https://github.com/hoglet67/RGBtoHDMI/wiki

For this to work proberly the component video signal is explicitely created in a high-contrast way to allow easy interpretations of the colors. 
This external RGBtoHDMI adapter can be used in a very flexible way to support a very wide range of vintage computer models.

Currently there is no purely internal solution available, but I guess it could be possible to modify the software for the Pi in a way to make it work with the digital signals that can be exposed by the component mod in some way. Here I would also have to modify the firmware for the mod a bit.

@IanSB
What do you think about such a purely digitial approach?  With the FPGA board producing 4 color bits + sync and also a 8MHz pixel clock (all already with 3.3V levels) this should be an easy task for the Pi. I don't know about auto-configuraiton, tough. Using this auto-configuration with the resistor is already used for the Amiga. Maybe the Pi could decide what to do based on the pixel clock frequency? This could also make for an easy Atari 800 integration.

mistepien, 2021-03-07:
I meant "no component middle man" because now it is doable with OSSC or RetroTink.

IanSB, 2021-03-07:
@mistepien 
>I meant "no component middle man" because now it is doable with OSSC or RetroTink.

Using the component output with the external version of RGBtoHDMI is not the same as OSSC or RetroTink. With RGBtoHDMI the output is used to transfer the signal in a digitally perfect way so that there is no analog quality loss and the output is bit identical to a direct digital connection.

IanSB, 2021-03-07:
@c0pperdragon 

>I don't know about auto-configuraiton, tough. Using this auto-configuration with the resistor is already used for the Amiga

It might be possible to use the two unused up and down button inputs to select 1 of 4 possible configurations. They are both pulled high by default so pulling one or both low could signal a different hardware configuration.

mistepien, 2022-01-27:
Ok, let's go that way. How can I connect external RGBtoHDMI upscale to the mod?