c0pperdragon/C64-Video-Enhancement #58: Using of external Sync is not Working

DerGali2020, 2021-03-10:
Hi,
first thanks for your awesome Project. I'm trying to use your board as RGB Output with a 8 PIN MiniDIN(using this pcb for easier install https://oshpark.com/shared_projects/0hrYbjDu ) , which all my consoles have and hopefully all homecomputers get :). I used the program for pallet switching to get it into RGB Mode. If i use RGB mode with ns (no sync) sync from Composite or LUMA IN / LUMA OUT from on the RFBOARD is not working. i get a sync but the screen stays black on my OSSC and Rad2X. If i use RGsB all is fine. I don't know if thats the cause but on RGB with external Sync my OSSC says 311P if i use RGsB my OSSC Syncs with 312p. Maybe this is the issue? I've bought the rf replacement board from VGP so i think it doesn't have the newest firmware.

Thanks in Advance
DG

c0pperdragon, 2021-03-10:
I not sure I completely understand what this 311P here means?  
Is this what the OSSC shows you when you hook up the original luma signal to it. Would be best if you could leave away all the RGB connections just for testing the sync.  
311P is super strange. Maybe the OSSC can not properly handle the sync the VIC-II produces
Do you happen to have an oscilloscope so you could test the signal from the C64 and see what is going on there?

But when you are already doing this permanent internal install, you could change the board a bit so it would also generate a clean sync signal for your custom output connector. When you do that you would then no longer have the sync available on the normal lines (as for YPbPr or RGsB), but have it on this connector exclusively.
If you want to do that I can give you instructions here.

DerGali2020, 2021-03-10:
311P is what the ossc Display shows if i hook up the original luma signal to it. Sorry i don't have a Oscilloscope but maybe i could borrow one, which will take a few weeks. I will try to remove the rgb lines to see if it changes anything. If you have enough time to tell me how to modify the board that would be awesome :).

c0pperdragon, 2021-03-10:
Ops!  I mixed up the description. The 470 Ohm needs to go to the pad of the R1.  The 100 Ohm connect to ground.