c0pperdragon/C64-Video-Enhancement #59: Upper corner's are a bit "stretched"?

olliraa, 2021-03-16:
Hi.

Been running this wonderful piece of hw for over a year now :) Now, as everything else is perfect, I started to wonder what might be causing the little "stretching" in the upper corners. It's not anything major, but as it cannot be corrected with the really extensive Sony monitor geometry settings, I wonder if it's something related to some setting in the video board? The same does not happen for example with ZX48 with the A-Videoboard. Both ZX48 and C64 
[image]
output RGsB (sync on green).

C64 250466, VICII R5
Fw 2.6 (yes, I know there are later ones ;)
New PSU
Sony BVM20F1E

c0pperdragon, 2021-03-16:
This kind of distortion can only be caused by the monitor itself. The pixel generation of the C64 is tied to the system clock and there is just no conceivable way how the video chip (and my mod) could cause different pixel rates on different places on the screen.

I have seen on many CRTs that the horizontal stretching depends on the overal brightness of the image. You could test if this also happens here by setting the border color to the same as the background, clear the screen and write a few vertical lines at the edges in questions.

olliraa, 2021-03-16:
> This kind of distortion can only be caused by the monitor itself. The pixel generation of the C64 is tied to the system clock and there is just no conceivable way how the video chip (and my mod) [...]

Ok, thanks cOpperdragon :) I'll try to adjust the brightness. How would I do the vertical lines stuff? Sorry, I'm no guru here :O

olliraa, 2021-03-16:
You'are a genius - but that's nothing new 👍  Lowering the brightness actually made a big difference. The upper corners are now perfect. I'd never thought of that 🥇 I guess adjusting the palette could have some impact on the perceived need to raise the monitor brightness, right?

closed 2022-02-11 by c0pperdragon