c0pperdragon/C64-Video-Enhancement #61: Snow effect

InsertDiskTwo, 2021-04-17:
I have noticed that in dark areas of the screen (but not total black) there is a faint snow effect going on. For example, in the game Witch Switch on the grey border (once you start the game). 

I tested with both a Retrotink x2 Pro and Ossc and got the same effect. It is much more visible with the Retrotinkx2 Pro.

c0pperdragon, 2021-04-17:
I had some reports of slight image noise previously. So far these were always caused a a noisy PSU. Maybe you have access to an alternate power supply to test this ?

InsertDiskTwo, 2021-04-23:
Hi c0pperdragon, sorry for the late reply. I do not have access to another PSU. I have had a friend coming over to check the PSU for noise. I am attaching measurements and pictures from scope below. What do you think? The problem is that we do not have anything to compare to. 
This is both power rails measured at the same time, the rails are not to scale.

3.3 volts right at the JTAG connector:
Period of the noise
[image]

Peak to peak of the noise
[image]

5 volt peak to peak at the board's regulator input:
[image]

This is the [PSU we are using](https://www.c64psu.com/c64psu/45-1379-commodore-64-C64-1541-fdd-power-supply-psu.html#/11-color-black/37-ac_cable-eu).

Thank you!

InsertDiskTwo, 2021-04-23:
The more we measure it the more I think it is the PSU. Here is the video signal with the 3v3 rail under it:
[image]

c0pperdragon, 2021-04-23:
Still it is hard to guess. Normally when the PSU is causing the issue, it it somehow time-related to the power input frequency. In PAL regions this is normally 50Hz, as is the vertical screen frequency (in NTSC regions, this is 60Hz instead).
As these frequencies do not match up perfectly, the visual distorion should slowly travel up or down the screen. If this is the case, the problem is surely the PSU.

InsertDiskTwo, 2021-04-23:
Thanks a lot! I will see if I can get my hands on an original PSU to see if it improves thing. Is there an aftermarket PSU I can get which you think won't have issues?

c0pperdragon, 2021-04-23:
I would not use an original PSU at all. These are really dangerous by now and can actually kill your machine.
Unfortunately I can also not give you any recommendation. 
My own PSU is a modded original one. I replaced the voltage regulator and added a heatsink and a fan. But again I can not recommend this to you as it is a bit dangerous.