c0pperdragon/C64-Video-Enhancement #64: C128 support (#2)

c0pperdragon, 2021-04-27:
This is a continuation of https://github.com/c0pperdragon/C64-Video-Enhancement/issues/4

lolof70, 2021-04-28:
Thank you very much for your great support ! Really happy, I will build one for my C64.
Thanks again.

lolof70, 2021-05-07:
After some more testing, I do not know why, but some times the image is stable, sometimes not.
When it's not stable, there is light movement mostly horizontally or lightly in diagonal.
Moving a bit down to up... Like if it was doing this with the back colours. The rest seems stable.
I thought it came from the PSU because the video cable connected from the fpga board to the A-board was very near to the psu in the 128DCR. I turned the cable the other way so it goes on the opposite side of the psu and it changes nothing.
The PSU has been recapped as all the 128DCR....

I read about a 47pf fix or a fix from #38 but I can't find where are the exact infos.

Is it something I could try. 

Very strange because it can do this for a moment then stop for some times  then start again....

c0pperdragon, 2021-05-07:
You should also try different output modes. You can switch between 240p/288p and 480p/512p with the mode selector.

lolof70, 2021-05-07:
Different output changed nothing, also tried with OSCC and the result was the same, even amplified.

I can't almost not believe it....

The dcr was plugged  on a 220V power multi block...
I tried powering the C128 alone from another wall plug. No more on multi block....

It's working a lot better. this effect is almost gone... Just time to time but nothing as before when it happened...

Strange. I checked on the normal video din output, this effect are not to see.

c0pperdragon, 2021-05-07:
Could you do some video capture with the PSU that causes the worse picture? I still can not quite visually imagine what you mean.

lolof70, 2021-05-08:
Yes, done

https://youtu.be/xooaiQYsORM

c0pperdragon, 2021-05-08:
So, at least I see that it is no problem on the digital side between the C64 and the mod board. The problem here is some analog noise.  

Your experiement with touching was very important. It is absolutely not normal that such minor capacitive changes should cause any noticable difference on the screen. My only conclusion is that there is something faulty with the DAC resistors or the soldering of the FPGA pins. Maybe one of the many legs of the FPGA has no proper contact or some flux residue causes leakage current between pins or something like that.

lolof70, 2021-05-08:
The board was put in a propanol ultra sonic bath. It's very clean.
I will check again the legs from the fpga. Which one are de DAC resistors ? 

Also... I build 2 board (I forgot this !!!!).. I will connect the second one and check if it does the same, also with touching...

lolof70, 2021-05-09:
The problem with touching the FPGA is solved.
You were right ! I had 2 pins not well soldered on the FPGA... I missed them when I checked after soldering...

I checked again on my big TV who can do 240P.... Best results is using 240P and with the power cable alone in a 220V plug.
Probably same as before. 480P is always with this noise on big and small TV...

I will try to feed the video on the small TV through HDMI with OSCC and 240P from FPGA....  The small TV is the one I want to use with the 128... If it does as well as on the big TV 240P, I will be happy :-)

c0pperdragon, 2021-05-09:
So, if there is at least one TV that shows the picture perfectly, then the signal is obviously perfect already and no action is needed on part of the YPbPr mod.

lolof70, 2021-05-09:
Yes, the mistake was the 2 pins not soldered on FPGA and a bad combination of TV during the tests.... 
I played already a lot with it, impressive with the right TV ! Thank you.

mincebert, 2021-07-29:
I've been watching this (and the previous thread) hungrily, in anticipation for my 128DCR...

Does anyone know of someone selling these boards for the 128?  My soldering skills aren't up to something like this, so I was hoping someone would offer them, as per the C64.

Thanks in advance for any solutions!

djcarlin, 2021-07-30:
I too am interested in purchasing a kit, for my C128 - similar to the C64 kit I bought from VGP.  Thanks!

c0pperdragon, 2021-07-31:
As it currently looks, the global semiconductor shortage also hits us hobbyists pretty hard. The FPGA that is the central part of the mod is sold out and may become available again earliest 2022-7 (but no guarantees, of course). Even then I am not sure if I will have the time to take up soldering these FPGA boards anyway.

There may still be another option. The FPGA board from the C64 mod is basically identical to the shrunken-down version for the C128, So If you want to become creative, you could actually order a C64 mod from videogameperfection.com, discard the C64-VIC-adapter (a shame to waste it, but, well...). Then add a C128-VIC-adapter for which parts should still be fairly easy to get.  Fitting the C64-FPGA board into the C128 is probably quite difficult and may require cutting away some shielding metal, but I guess you could do that.
If you want to go that route, I could check if I can get all the parts for the adapter (still have 6 PCBs for that) and build these for a reasonable price.

c0pperdragon, 2021-07-31:
You could actually ask user @bodgit if he has some kits remaining from his batch of 5 he made a few months ago. (https://github.com/c0pperdragon/C64-Video-Enhancement/issues/4).

djcarlin, 2021-08-01:
Ideally, VGP could start carrying the C128 kit once the FPGA is available again in coming years?   Matt @BuckoA511 expressed interest in carrying such a kit here:

https://github.com/c0pperdragon/C64-Video-Enhancement/issues/4#issuecomment-773215229

bodgit, 2021-08-02:
Yes, I do have some spares still. I will email both @mincebert & @djcarlin to the address that is in your profile or commit history.

mincebert, 2021-08-03:
> Yes, I do have some spares still. I will email both @mincebert & @djcarlin to the address that is in your profile or commit history.

Thanks — that's much appreciated!  (I don't think I was showing an email in my profile or if there was another way to contact me, but I think I am now.)

bodgit, 2021-08-04:
> > Yes, I do have some spares still. I will email both @mincebert & @djcarlin to the address that is in your profile or commit history.
> 
> Thanks — that's much appreciated! (I don't think I was showing an email in my profile or if there was another way to contact me, but I think I am now.)

You hadn't, but I cloned one of your repos and used the email address you used for commits, it went to rcf@m****.net.

bernouilli, 2021-08-06:
Hello,
I just found the great project and would like to get a kit for my c128.
Does anyone have any kit left?
Regards.

bodgit, 2021-08-07:
I have my last two left now. From the old #4 issue I think @nickgoodmanuk wanted one as well so I'm tagging him as well. If you can let me know your email addresses (I can't see an address in your profile) I will contact you both with details.

bernouilli, 2021-08-07:
That's perfect. Here is my email address : bernouilli92@hotmail.com

nickgoodmanuk, 2021-08-10:
Brilliant ! My email is nick@nickgoodman.plus.com.

Many Thanks.

nickgoodmanuk, 2021-08-17:
> I have my last two left now. From the old #4 issue I think @nickgoodmanuk wanted one as well so I'm tagging him as well. If you can let me know your email addresses (I can't see an address in your profile) I will contact you both with details.

I'm another happy C128 component video customer !   Many thanks to @bodgit for building and sending me the boards and obviously @c0pperdragon for his design.   Fantastic picture!

[image]

[image]

[image]

[image]

[image]

c0pperdragon, 2021-08-17:
Voll gerne  :-)

mincebert, 2021-08-22:
I've now fitted mine from @bodgit and it works brilliantly in my 128DCR.

[image]

I thought I'd attach some pictures for reference, to show how I moved the capacitor adjacent to the VIC-IIe socket by putting some short wires in, to lay it behind the other one.

[image]

I then sawed the RF shield / heatsink cover in half so it just covers the 8568 VDC side.  I need to get some stick-on heatsinks for the VIC-IIe.

[image]

The only minor issue is the IDC connector into the CPLD board is _really_ tight up against the PSU tray: I have to connect the cable first, then drop that side of the PSU in and then press gently to get it to slot in on the other side.  This bends the socket a fraction of a millimetre:

[image]

[image]

So, if there was an iteration of this, it would be good to shift the pins for the IDC connector on the CPLD board a little bit away from the PSU tray — probably 2-3mm would do it, but a little bit more would be even better.

I don't know if anyone else installing this has a DCR (or even a D) and had the same issue: the other ones I saw here were flat ones (I think @bodgit said his was a rare 128CR), so presumably they don't have this problem.

mincebert, 2021-09-02:
Brief follow up on this, now I'm back from holiday and some final goodies have arrived...

I got a heatsink meant for an M2 SSD 70x22x6mm and it fits beautifully over the VIC-IIe and my multimeter's temperature probe shows the temperature around 36-37ºC on the middle (where I was getting 46+ºC on the bare chip):

[image]

I also made up a TRRS-IDC cable to connect the board to a Raspberry Pi RGBtoHDMI with analogue board and YUV mode loaded onto the CPLD and a jumper across pins 9-10 on the JTAG connector to replace 576p scanline mode with the 288p high contrast palette:

[image]

This is giving a beautiful image that you could forgive for being an emulator (this captured using the RGBtoHDMI directly, so it's a pure PNG of the source image):

[image]

Thanks everyone — this is absolutely amazing: I've gone from a flakey S-Video image with noise and flickery colours to a beautiful HDMI picture.

rasivert, 2021-10-13:
Oh, I see. Well I guess patience is the word here.

c0pperdragon, 2021-10-30:
As the content of the picture looks mainly OK, I would guess thath the connection between the C64 and the FPGA is working fine. So the fault probably lies in the video signal generation. I would guess that the csync signal is not mixed correctly into the Y output. With an oscilloscope you could easily check this. 
Otherwise check the signal path for the csync signal from pin 46 on the FPGA through R1 and see if all outputs of R1 - R6 are connected together.

honselio, 2021-10-30:
Oha, checked R1, 130 Ohm instead of 1,1k Ohm, an angström solder bridge was the culprit. Board 2 is working. Vielen Dank und Grüße ins schöne Wien!

Braincell1973, 2021-11-04:
Hi , are these boards available ? Or is there gerbers to have some made ??

c0pperdragon, 2021-11-04:
The install consists of two parts which (for historical reasons live in different repos):
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/c128board
https://github.com/c0pperdragon/C64-Video-Enhancement/tree/master/smallreplacement
So you can have your own PCBs made. But due to the global semiconductor shortage, the FPGA needed is not available (and probably will not be for a year or more). 

videogameperfection.com seems to have some C64 kits in stock. You could use this by replacing its VIC-II adapter with your home-made VIC-IIe adapter (link above) to fit the C128.  Also you need to find a way to fit the larger FPGA board into the smaller space of the C128. If this is too cumbersome, you will have to wait for the global delivery chains to fill up again...

kavanozz, 2021-11-23:
Using 2 of the 24 pin long tail sockets for VIC-IIe adapter would be easier and cost the same if you don't already have 40 pin. No cutting required.
https://www.mouser.com/ProductDetail/Mill-Max/111-93-624-41-001000?qs=WZeyYeqMOWdZSxh1h66G8A%3D%3D
https://www.digikey.com/en/products/detail/mill-max-manufacturing-corp/111-93-624-41-001000/260654

djcarlin, 2021-12-23:
I finally got around to installing the C128 kit @bodgit sold me (thank you!!) and its working well.  I did have one question about s-video output.  I noticed it seems to have issues with the color "Red".  Top photo is S-VIdeo on Dell 2007FPb via OSSC & Koryuu:

[image]

Second photo is same display, same OSSC - but no Koryuu -  using component output of C128 instead:

[image]

c0pperdragon, 2021-12-23:
The questionable quality of the original VIC signal (composite and s-video on the original A/V port) is the main reason to create component video from the digitial communication signals. 
The mod provides a high-quality video amplifier to make the best of the analog VIC output  to give you the best s-video and composite it can. I am quite surprised how well the image actually looks. 
Of course it can not reach the quality of the independently created component video signal.

djcarlin, 2021-12-23:
I dug out a 64C shortboard and Red on Dark Gray looks just as bad via s-video.  

I love having emulator level picture quality on real hardware !  Thank You!

ami-commo-fan, 2021-12-27:
> I don't have any more left now, sorry. It might be a while before someone builds some more as last time I checked, the FPGA used has a very long lead time on it, (April 2022 I think).

I would take one as well when bodgit builds some more.

rasivert, 2021-12-28:
> > I don't have any more left now, sorry. It might be a while before someone builds some more as last time I checked, the FPGA used has a very long lead time on it, (April 2022 I think).
> 
> I would take one as well when bodgit builds some more.

Same here!

themaddman, 2022-05-29:
> > Very interested in this, are there more available?
> 
> I don't have any more left now, sorry. It might be a while before someone builds some more as last time I checked, the FPGA used has a very long lead time on it, (April 2022 I think).

Add me to the interested list, if the FPGA ever becomes available again.

robby-d, 2022-06-07:
I will take 2 of these when they become available again.

johansmolinski, 2022-09-17:
I'd love a board compatible with my C128DCR as well. Please sign me up if there is a list for that anywhere. :)

bodgit, 2022-09-29:
They're charging between $75-$90 per chip (depending on quantity). They normally would be less than $10. That's quite a markup!

I think they are being produced, but they're being snapped up by volume buyers so there's none spare for people like us who are essentially hobbyists. I think I've purchased as many as 10 in one order, which is nothing really.

johansmolinski, 2022-09-29:
Ok, I had no idea they originally were that cheap. Heck, an FPGA for less than $10?! Wow...

Do you have all other parts for what is needed to build a board that will fit inside a C128DCR?

thepebbler, 2023-04-15:
> They're charging between $75-$90 per chip (depending on quantity). They normally would be less than $10. That's quite a markup!
>
> I think they are being produced, but they're being snapped up by [...]

I've just grabbed two C128's (A C128D and a regular C128) , both are broken, but the plan is to get the C128D resurrected and wanted to get a the small board version, but don't have the kit to solder the FPGAs on!

Is anyone planning on doing a batch? I might be able to help source the FPGA if that is the blocker, I can ask in work and get our CEM team (they work with our CEMs making our PCBAs) to see if they can source the FPGA at a reasonable price, and get some at cost.

scorpio-ny, 2023-05-09:
Before I order a PCB, I wanted to check here to see  if anyone has an extra VIC-IIE PCB adapter or an extra assembled VIC II-E adapter they do not need and would be willing to sell? A while ago I had purchased a 64 kit from VGP in which I was saving for when I would get a C128 (finally acquired). Thank you!

johansmolinski, 2023-05-31:
I am offering to trade a fully working C128 PAL with 6581, healthy yellowed, **without** PSU for a full c0pperdragon C128 kit that will fit inside my C128DCR. I can do whatever needed to fit it, if it means cutting things and relocating capacitors or whatnot, as long as the kit is complete. I live in Sweden, so I would prefer to do it within EU, but everything is possible.

c0pperdragon, 2023-05-31:
c0pperdragon here.
Even I don't have any of those boards. But maybe you would be interested in
my follow-up product. It solves the same issue in a new way using a small
mod board and an external RGBtoHDMI. See https://github.com/c0pperdragon
for the C64 variant.
If there is demand, I could also make a C128 version.

johansmolinski, 2023-05-31:
Thanks for prompt reply! Your link is to your profile. Which of the repositories are you referring to? And will it be usable for hooking up the C128 to a 15 kHz CRT on RGB? That's all I care about. :)

c0pperdragon, 2023-05-31:
it is the wiki section of github.io  .
and currently the solution is for HDMI upscaling.
but maybe i can dream up some DIY circuit to get RGB again.

rasivert, 2023-05-31:
> I could also make a C128 version. 

Very much interested in a C128 version!

johansmolinski, 2023-05-31:
Would the RGB2HDMI be able to output 15 kHz, like "direct video" on MiSTer? If so, you could probably use a cheap HDMI to VGA adapter to get good 15 kHz RGB that way.

Still I don't find the information you are referring to. I would really appreciate a direkt link, please. :)

c0pperdragon, 2023-05-31:
https://github.com/c0pperdragon/c0pperdragon.github.io/wiki/VIC-II-dizer

c0pperdragon, 2023-05-31:
I hope, I do not give away confident information, but videogameperfection.com seems to be in the process to stocking up on the C64 variant of the mod.
Looking at the pictures of the C128DCR motherboard, at least the FPGA half of the mod may actually just fit.
Maybe someone has/can make a VIC-IIe pickup board. 

In case you can not get a suitable pickup board, you can even try a super-badass hack, I just came up with: Instead of putting a pickup-board between the VIC and its socket (which will only work when you have the correct board), you could instead piggy-pack the C64 pickup board onto the VIC. Most signals are at the same place, so you only need to cut away 6 of the pins and re-reoute these signals . If you want to go that way, I can give more detailed instructions.

c0pperdragon, 2023-06-01:
I created a new thread for this topic, at this here is already much too long again:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/89

closed 2023-06-01 by c0pperdragon