c0pperdragon/C64-Video-Enhancement #65: support for C64 Component Mod

tronarea, 2021-04-27:
I install the fpga board and the VIC II Adapter, i replace the c64 VIC II ic holder and replace with a drilled 
precision IC holder and the picture its generate was perfect wonderfull!!!
 
But on jump and run example giana sister the green woods change from green to white also on x-out sprites.
 
My setup c64 with vic II you can see the model also on pictures.
 
 
What can i do, i love the modul its so clear picture but with the grafic errors i cant use it so.
 
Im a electronic specalist with good skills i know all about esd and so on i was realy carrfully with this board.
 
Thanks for your time i hope you can help

Best regards Jürgen 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

c0pperdragon, 2021-04-27:
This is a pretty unusal effect. I looks like exactly this 8 pixel height stripe near the bottom of the screen is affected. For me this looks a bit like if the color ram of the C64 is somehow misbehaving.   
Could you check this by doing a direct comparision with the original video signal from the A/V port?

tronarea, 2021-04-28:
First of all, many thanks for the quick response!

That is a good idea, unfortunately I have no connection option.

then tried to connect the video out via scart to the ossc scandoubler, unfortunately without success.

then processed the c64 chips with cold spray as well as with heat gun.

One was sensitive. I will desolder the vram from an existing device and install it in mine.

For your good idea you have to
do I have to fall back on a colleague who still has a tv with a connection. he is currently on vacation so it will not be possible to continue until next week.

thanks for the idea, I'll get in touch with the results next week.

have a good time and see you soon

best regards

jürgen

tronarea, 2021-05-05:
Hello again, 

on the video out also have the video problem see attached photo

I think it's a problem on my c64 the multi chip 251715-01 is maybe defect.

thanks so much for your time

I think I must replace my c64? 

have a good time

best regards
Jürgen
[image]

closed 2022-02-11 by c0pperdragon