c0pperdragon/C64-Video-Enhancement #66: Will it fit 250407 rev.B

magazaki, 2021-05-19:
Hi

My board is 
Commodore 64
1983
Assy No. 250407
Artwork no.251137 Rev.B

VIC chip:
MOS 6569r1
2381
40 pins
12Volts

Seems like it could be an issue fitting the adapter, possible conflict with adjustable resistor and capacitor?
[image]

c0pperdragon, 2021-05-20:
There are a few difficulties to overcome with this specific board. One user made a nice video using this:
https://www.youtube.com/watch?v=oo_gWjK-gGw&t=534s
He also uses an additional hardware to then convert the output signal to HDMI, but this is optional of course.

magazaki, 2021-05-20:
Great, thanks alot! I will give this a try!

magazaki, 2021-05-20:
:)

closed 2021-05-20 by magazaki