c0pperdragon/C64-Video-Enhancement #67: Sync but black screen

magazaki, 2021-05-27:
Hi

I aquired this board from videogameperfection.com
I have followed the info on [https://youtu.be/oo_gWjK-gGw](url)

I also have the OSSC, it shows sync on YPbr, 316-p, 31.21kHz 98.77Hz
But the screen remains black.
My C64 Board is 250407 Rev B and my VIC chip is MOS 6569r1
I notice is that the VIC is not seated very firmly in the roundholed adapter from this kit.
What can be done with this?

I have shorted the JPLUM1 bridge as described in the Wiki (allthough I had applied power once before I did this...is that very bad?)

So I am a little stuck and don't know what to test next.
Any ideas is very welcome!

thanks
brg,
Magnus AKA Magazaki

magazaki, 2021-05-27:
Update: it's a bad connection issue on the VIC and adapter board. I just got it working by pushing it down gently while powered!
Any tips on how to get the flat VIC pins to make a good connection in the round holed adapter socket would be much appreciated 😀

magazaki, 2021-05-27:
Brute force was the answer to my socket contact issue. I guess I am just not familiar with machined sockets

c0pperdragon, 2021-05-27:
This was actually one of my favourite issue threads. Raising the problem, working through the solution and closing the issue in 2 hours all by yourself  ;-)