c0pperdragon/C64-Video-Enhancement #68: Alternative part for Altera 10M02SCE144C8G?

modeler, 2021-06-09:
Presumably due to the global chip shortage, the FPGAs listed in the BOM are showing as unavailable everywhere with long lead times. Mouser does not expect any significant stock of these until May 2022!

Could there be an alternative FPGA with the same footprint or is this project effectively stranded? Any advice would be most appreciated. Thanks in advance!

c0pperdragon, 2021-06-09:
That is a real problem right now. Not only this specific part, but basically the whole chip family is sold out. It seems there are some remaining parts with 16k gates upwards (instead of the 2k), but they cost much more and are not immediately compatible. You would need to modify the PCB to make them fit. Also you need to adjust the firmware to the changed pinout.  All this is quite a major task and not cheap either (€44 instead of €6 for the FPGA alone).

c0pperdragon, 2021-06-09:
If you are quick, you may get the last assembled board that is available from https://videogameperfection.com/products/commodore-component-video/

modeler, 2021-06-09:
Thanks for the reply c0pperdragon. Unfortunately VGP don't ship to the UK due to Brexit issues, I have been talking to Matt on Twitter about this.

modeler, 2021-09-23:
Quick update, VGP resolved their issue with UK customs and are shipping again. I was able to source a mod board and fitted it a few weeks ago, it is amazing. I would love to build some of these myself, I had the PCBs made and sourced all the parts... except the actual FPGAs. I really don't know when we're likely to find them for sale again.

[image]

c0pperdragon, 2021-09-24:
Sorry, there is basically no replacement that fits the current PCB design. At mouser there are not even the more expensive parts that could be made to fit by slightly redesigning the PCB. 
I guess we have the same problem here as the global automotive industry...

modeler, 2021-09-29:
Please don't apologise, this is not in any way your fault. I hope that at some point before VGP's stock runs out, we will all be able to source some of these FPGAs.

closed 2021-09-29 by modeler