c0pperdragon/C64-Video-Enhancement #69: Firmware source code?

mwedmark, 2021-06-14:
Is the firmware from https://github.com/c0pperdragon/A-VideoBoard.git usable directly out of the box? I don't understand where the code to build the Quartus project for this board is.. Thanks in advance, great work! Sorry to see shortage of circuits puts eveything to a halt, but lets it's getting better.

c0pperdragon, 2021-06-15:
Ho to the subfolder for

c0pperdragon, 2021-06-15:
Should not use github from a mobile... 
Anyway, go to the subfolder for the C64 mod. Thetr is the Quartier project.

mwedmark, 2021-06-15:
Thank you! I got 95% usage when building it in latest Quartus, that is a great Device/Project match! :-)

closed 2021-06-15 by mwedmark

c0pperdragon, 2021-06-15:
Yes it is a tight fit. Had to restructure the firmware a few times along the way to stay inside the bounds.