c0pperdragon/C64-Video-Enhancement #71: No C64 screen, only test palette output

Syntec1, 2021-09-07:
Hi,

I feel kinda stupid as I get the feeling I'm missing something here.
So the issue is that the Vic2 board has been connected and everything to the FPGA board which I assembled and is outputting the default palette colors.
I have clear output on the Svideo out and programmed the FPGA with the C64mod file.
As I can see the Vic2 board is outputting signals to the FPGA board so I'm guessing that I did something wrong with writing the software to the MAX10 as that is new for me?
[image]

Can someone assist me for troubleshooting the issue?

Thanks!

c0pperdragon, 2021-09-08:
You have programmed the test firmware to the board, which is intended to check the only the video output output circuitry (which seems to work perfectly).
What you now need instead is the actual firmware for the VIC2 emulation which you can get here: https://github.com/c0pperdragon/C64-Video-Enhancement/releases

Syntec1, 2021-09-08:
I have installed the 2.10 firmware and I get an input signal with only a black screen.
I double checked the svideo output and there I get a c64 screen.
[image]
I have a 250407 board and I read a lot about the clock issue, do I need to do this or is there another fault?
The only thing I just noticed is that I installed for the 3 resistors (R5,R10 and R15) that are 8K, the 8K2 resistors, will that have a to big of an impact on the signal or as the output circuit is working should this be okayish?
Just want to eliminate anything that could cause the issue. (For some reason I got 8K2 marked as 8K after measuring and checking)

c0pperdragon, 2021-09-08:
When the test signal did work fine you have no problems with any of the resistors of the signal output circuitry. 
The problem is very likely that the signals (at least the clock) from the VIC2 do not reach the FPGA. If you have an oscilloscope, probe the various signal paths the check if there is activity. Otherwise do very close visual inspection of all the pins. Especially the input pins of the FPGA can cause trouble with solder bridges.

Syntec1, 2021-09-14:
Excuse me for the delay, my VicII 6569 decided to quit on me and I replaced it with a 8565 R2 and made a mod in the 12V pin13 line so it's now 5V and had the oscilloscope ready then to my surprise your board started to work.
I also changed your board to 5V Vic instead of 12V just in case.

[image]

The edges and text I assume will be the pixel clock issue were I need to do the mod right?
Just to check with you

Syntec1, 2021-09-15:
I had a delay in testing as my original VicII suddenly died on me 6569.
I decided to replace it with a 8565 R3 and mod the 12v voltage line to 5V, this works, I changed the FPGA board also from 12V to 5V Vic just in case.
After that modification I was ready with the oscilloscope to start measuring but the board started to work.
Noticed I needed to do the pixel clock modification which I did and I have an almost perfect picture except for a bar moving slowly upwards.
[image]

You see the distortion on the edge which is moving slowly upwards, any ideas were that can come from?

c0pperdragon, 2021-09-15:
When the effects moves slowly up or down the screen, this is a very good indicator that it has something to do with the 50Hz power grid frequency. The issue can either be the C64s (including the mod board) or he Monitor. Do you have some means to check how it looks on a different monitor/TV?  
And if this issue is independent of the monitor, how does it compare with using the original s-video signal?

Syntec1, 2021-09-17:
It was the upscaler I was using, now connected to the RGBtoHDMI it is perfect.
Thanks for the support, very fun project to build from start to finish!

closed 2021-09-24 by c0pperdragon