c0pperdragon/C64-Video-Enhancement #73: Need help with FPGA Mod + external RGB2HDMI

lolof70, 2021-11-02:
Hi,

I do not know if I have to post this here or to the rgb2hdmi page..
I am trying here. I just added the external RGB2HDMI to the FPGA mod in a C64.
FPGA mod is working perfectly...
I have problem with the RGB2HDMI mod :-(
See the pictures. Colours are not good. I have the last version on the FPGA board. So I did not had to programm the C64 palette.. I put a bridge on the pin 9&10 from the fpga jtag connector. The switch is on the right side (facing the C64).
My cable seems to be OK, I checked it again now..

I am afraid it could be a problem with the analog board issue 3... I do not know where to look.

Could someone help me. Thank you.

[image]
[image]

c0pperdragon, 2021-11-02:
When the mod board gives correct colors in normal operation (no jumper), then it is very unlikely that this is the problem.
So the issue is either with the cabling (if you are lucky) or with the analog board (unlucky) or it is just a matter of correctly calibrating the input voltage levels (extremely lucky). 
Try playing with the rgb2hdmi settings first and see if you can make the board discriminate 3 different input levels on each of the signal components (Y, Pb, Pr). The sync already is detected perfectly. 
I don't remember how to use the settings menu of the RGB2HDMI but after pressing the button, you should be able to guess your way through the menu.

lolof70, 2021-11-02:
It's like it's missing red and blue colours... I saw yellow, black, braun,....
I re-checked the cable. I do not see a mistake...
I changed the only parts coming from ebay china with another from the lot I bought, just in case... Same result.
I went in Menu and tried to change what I could, like contarst etc....  Dors not help.
I also connect all on another monitor. Same....

How do I calibrate the output voltage ? I did another board, digital for C128 and this one worked directly with default setting on the same monitor...

lolof70, 2021-11-02:
Just found an error I did on the analog board......
Maybe it's the problem.

R1 should be 470 ohm... I do not know why, I put one of 100 ohm....

lolof70, 2021-11-03:
Tried with R1 470 Ohm on the analog board. No difference...

A last question, on my FPGA board, I replace R18 and R19 by 75 Ohm, should I put the original value one ?
Can this be a problem ?

I also put my cable connection here just in case......  If I well understood the wiki, this should be correctly connected...

[image]

c0pperdragon, 2021-11-03:
Yes, this would definitely make a difference. With this lower-value resistors the output levels for Pb ad Pr would be muhc lower. This would lead to less saturated colors when used with a normal YPbPr-enabled TV. But due to the the way the RGB2HDMI reads the analog signals, whis would very likely cause the input to be incorrectly interpreted when using the default settings. 
So either replace the resistors again or reconfigure the input voltage levels for the red and blue lines using the RGB2HDMI's settings menu.

c0pperdragon, 2021-11-03:
Have a look at
https://github.com/hoglet67/RGBtoHDMI/wiki/Reference-Guide
in the section Sampling Menu (6 BIT RGB analog inputs)
This is where you can reconfigure the analog voltage levels.

lolof70, 2021-11-03:
Problem fixed...

Yes, it was only the 2 resistor I replaced. I put the original value back on the board and now, all is working perfectly !

Thank you for all.

closed 2021-11-04 by lolof70