c0pperdragon/C64-Video-Enhancement #74: C64RGB RGBns Scart support

rapiqui, 2021-11-14:
Hi,

I have installed a C64RGB mod on a C64C with success, being able to get a wonderful and crisp image from the TRSS YPbPr output on a LCD TV with composite video.

The fact is that I wanted to get rid of the TRSS connector and went ahead building a custom din8 connector on the C64 motherboard, getting only GND, Audio and Luma from the board, and freeing the rest of connections to accomodate R,G and B channels from the C64RGB mod.

Having tried the colodore palette (RGBns) and the original RGBns from Nojo-horem app I am unable to get the same colors on the RGB-Scart connection.

Please, check out an example image:

This is what I obtain from the RGB scart:
[image]

And this is how it looks on the component output:

[image]

Is there anything I should take in consideration?
Thank you

c0pperdragon, 2021-11-14:
Looking at your image is would say that the original blue channel is missing completely and the original red channel is fed into the output blue instead. This is true only if the blue text is really static.

c0pperdragon, 2021-11-14:
If the blue text is misleading, it could be just as well that red and blue are just crossed out.

rapiqui, 2021-11-14:
Thanks for your quick response.
I also thought the same initially, hence I tried to change the color channels connection to the scart, but this made some other weird color scheme that wasn't correct (red tinted).
Right now, testing with a multimeter, there is continuity from:

- Component blue RCA inner tip to scart pin 7
- Component red RCA inner tip to scart pin 15
- Component green RCA inner tip to scart pin 11

c0pperdragon, 2021-11-14:
Did you also connect all the relevant ground pins in the scart plug (5, 9, 13)? Leaving them floating could cause unexpected behaviour.
If yes, then the only thing I can imagine is a faulty scart input. Do you have a way to test this with another signal source?

rapiqui, 2021-11-14:
Hi,

Yes, ground pins are tied together and properly connected.
I have done a little bit more of testing and got some updates.
I connected the c64 with c64rgb mod on 240p switch position to an ossc instead of the lcd tv's scart and tried the AV1 with RGsB profile.  The OSSC displayed 312p, 15.63Khz and 50.10Hz, but no image was shown on the TV.
I switched then to 480p, the OSSC displayed 624p, 31.28Khz and 50.13Hz and the same incorrect color scheme appeared.
Finally I switched to AV1 YPbPr OSSC profile and then magic, the colors were perfectly displayed on the TV, so it seems that for some reason, even after having changed the palette to RGBns, the color signal is still being sent as YPbPr.

I did follow the instructions of nojo-horem's tool to change the palette, and after saving it, even power cycle the C64.

rapiqui, 2021-11-14:
Got it! It's currently working fine. Seems that the program could not upload the palette properly because I didn't switch the toggle to 480p before uploading the new palette. Once done I was able to power cycle down the C64 and connect it to the LCD tv via scart and using the switch 240p position the image colours are shown as they should. So, it was my fault.

Thank you anyway for your help and thanks for designing this beautiful piece of hardware.

closed 2021-11-14 by rapiqui