c0pperdragon/C64-Video-Enhancement #75: Troubleshooting help

mwardy1972, 2021-11-21:
I’ve had the component video mod from videogameperfection.com for a couple of months now and it's been working great ! But just the other day I turned on my c64 and the component output has stopped working.  The composite video output works fine. I just get a blank screen from the component output. Also, after a few moments the TV says ‘no signal’ when using the component output. I tried another tv in the house, same result. Unplugged the vic adapter board and ribbon cable, re-seated to be sure, still no go. A quick check with the multimeter, the voltage regs are putting out spot on 3.3V. I’m at a bit of a loss now trying to trouble shoot. I don’t own a scope or anything. Any ideas how to fix this ? Thankyou !

c0pperdragon, 2021-11-21:
If it suddenly stopped working this is probably a hardware fault. I don't think re-installing the firmware will do any good.  
This now may sound stupid, but have you possibly changed the position of the output mode switch to a mode your TV can not display? 

Now seriously: Troubleshooting the device without a means to see the signals is pretty impossible. As you have no oscilloscope we need to get creative here. You can use your working composite output image and super-impose any digital signal by coupling the signal into the analog video output.
Try this: Use as simple resistor with about 4.8K and feed the signal you wand to proble through this resistor into the junction point of R30,R31,R32 on the FPGA board. The best access to this point is at the side of R32 that is pointed towards the video amplifier AMP2.
To see if this rig works, start by probing pin 17 on the VIC-II itself, which is the system clock.  You should see very distinct horizontal stripes (4 pixels light, 4 pixesl dark) on your video display.

If this works, try probing all the signals that come trough the ribbon cable from the adapter board. All pins besides 15 and 16 should have at least some activity when in the C64 start screen. 
Please report back then.

mwardy1972, 2021-11-27:
Hi ! thanks for the help, I really appreciate it.  So, using your guidance and a bit of troubleshooting myself, I have found that U2, the 3.3V voltage regulator on the VIC adapter board is not functioning.  

As you suggested, I got a nice strong clock signal as you on pin 17.  But I didn't have anything on the ribbon cable at all.  I probed onto the VIC-II on the equivalent pins (i.e. DB0-DB11, A0-A5) and got the activity you described.  Which led me to review the schematic for the adapter board and I had missed this voltage regulator. I checked and found pin 5 had no voltage.  So to test it, I placed a bridge over to the output of REG 2, and ..... YES !! it works.  Found the problem, U2 is dead.  Now I just have to work out how to replace it, it is tiny.  Thankyou.

c0pperdragon, 2021-11-27:
My congratulations! Nice to hear that this impromptu oscilloscope actually worked. 

The voltage regulator on the adapter board is indeed pretty tiny. If you don't mind a bit of a mess and the damaged regulator is not interfering otherwise, you could leave it in and just wire a larger regulator to the capacitors directly.

mwardy1972, 2021-12-07:
I've managed to do a fairly neat job using a MCP1700-3302E 3.3V regulator.  Working great.  Might be helpul if anyone else experiences this issue.
[image]

closed 2022-02-11 by c0pperdragon