c0pperdragon/C64-Video-Enhancement #76: Help with making a RGB SCART cable for Sony CRT TV

tyristori, 2021-12-02:
I'm trying to make a RGB SCART for my Sony KV29K5E. Here is the SCART connector on the TV from the service manual.

[image]

I've connected signals from TRRS to SCART. Red to pin 15, Green to pin 11 and Blue to pin 7. And respective grounds to pins 13, 9 and 5.

I've set the output to RGsB with the palette editor and manually set the TV to RGB mode.. TV doesn't see any signal coming in. Screen stays black.

I also tried to connect composite video signal from C64 video port to pin 20 and ground to pin 18. And set the output with palette editor to RGBns. It didn't make a difference.

Worth noting that I can't get even composite signal to work with SCART. It does work if I connect composite signal to composite input on the front of the TV.

RGB from my Amiga works fine so the SCART input on the TV is ok.

Any help would be appreciated.

c0pperdragon, 2021-12-03:
RGsB will of course not help you, so RGBns is the right choice here with using composite from the A/V port for sync. So far you seem to have done everything right. But to get at least any picture at all it is pretty irrelevant which colors you have programmed. 

First thing is to solve the problem, that you can not get the original composite signal into your TV via the scart port. Without this, there is no sync signal and no hope the get any picture. For this test you need to revert the settings of your TV back to composite input (otherwise you may have some stable sync running, but the video may still be black). 

Maybe you can even do an investigation on how your amiga cable is actually wired up.

tyristori, 2021-12-06:
Success!

[image]

The RGB enable signal was needed. Picture quality is amazing on this big Trinitron.

Now I need to make a cable that takes the power for RGB enable from C64 video port.

closed 2021-12-06 by tyristori

Icelvlan88, 2021-12-06:
Hello,

I got a custom component cable made but can you give a small guide on how you did this? Looking to use RGB instead of component.

Closed #76<https://github.com/c0pperdragon/C64-Video-Enhancement/issues/76>.