c0pperdragon/C64-Video-Enhancement #77: CMOS or NMOS?

pushead, 2021-12-23:
Hi,

Hope I can get some advice from you:)

I'm going to install the RF modulator replacement for my C64C. But although this is a C64C is does have the longboard (assy 250466) and have the nice VIC-II 6569R5. But I'm not quite sure if this is a NMOS or CMOS? I beleive I need to know this in order to solder the JPLUM1 or not?

All the best!

laubzega, 2021-12-23:
It's NMOS.

c0pperdragon, 2021-12-23:
The long boards run with the higher voltages (all VICs with a number starting with a "6"). So you need to close the solder bridge in order to get propper luma levels. If you don't do this, nothing bad will happen, only giving you unusual luminance intervals on the image.

pushead, 2021-12-23:
Ah, I see! Thanks for helping me!

Have a nice holiday!

closed 2021-12-23 by c0pperdragon