c0pperdragon/C64-Video-Enhancement #78: Any ideas what could be causing these horizontal lines?

pushead, 2022-01-18:
Hi,

I've installed the RF-replacement (analog only). But I see that the picture is a bit distorted. If you see the pictures attached you'll see that there are some re-occuring horizontal lines which are most visible when the screen is dark blue/light blue (default C64 screen). It's not so visible when other colors are displayed.

My TV only support Composite by SCART unfortunately. So I can't test properly with S-video.

Any ideas what could cause this? 

BR,
Arne

Strange horizontal lines:

[image]

Looks ok:

[image]

Looks ok:
[image]

ikrananka, 2022-01-18:
This looks like it is likely to be related to Issue #38 and Issue #53.  If so, the solution is a timing mod which is board dependent.  Best to read through those two issues and watch the videos, in particular this one [https://youtu.be/oo_gWjK-gGw?t=2232](https://youtu.be/oo_gWjK-gGw?t=2232).

pushead, 2022-01-18:
> If I inderstand this correctly, you are just using the original analog
> output of the VIC-II here (with a simple analog-only amplifier board?).
> When you really have to use the composite output, there is not much you can
> do to improve the matter any further.
>
> The way composite modulates the color information into the luminance
> creates a signal that can not be perfectly seperated. This is most
> noticable on high-contrast edges, where the TV can not really guess wheter
> this signal is meant as a change of brightness or color. This is the reason
> why s-video was invented and was later superseeded by component video.

Ok! Thanks for feedback!

I've ordered a S-video cable also for my C64 to check if that helps. Doing
some testing on another TV:-)!

Thanks!