c0pperdragon/C64-Video-Enhancement #80: Flashing the firmware

c0pperdragon, 2022-02-11:
Recently I got some requests about problems with upgrading the firmware using an USB-Blaster. Instead of using e-mails I would like to collect the relevant information into this thread. Maybe this could be of some use to other users as well, so please share your experiences here.

To get a common baseline I will describe my current procedure for flashing the firmware. 
1. Wire up the USB-Blaster to your computer and to the FPGA board (check cable orientation).
2. Power up the computer or else make sure that the board has power.
3. Open the quartus programmer tool (my version is Quartus Prime 18.1, Lite). 
4. Drop the latest firmware (c64mod_2_10.pof) into the window or open it via the file menu.
5. Check  the boxes in the program/configure column
6. press "Start"

When the programming does not immediately abort, this means that there is at least a working connection to the USB-Blaster and the FPGA. I could also force an error-abort at 4% by selecting the checkboxes in the "Blank-Check" column. 

Use this image as a reference of what I am doing:
[image]

amerika13, 2022-02-11:
Thanks for making this. I will add in my process and pictures.

Process:
1. USB Blaster plugged into a Windows PC (tried two PC's on 10 and a laptop on 11). Windows recognizes it as I installed the driver (some difficulty there on a couple PC's). I plug the JST onto the pins (correctly via the lines on the board where the notch should go).
2. C64 VE board is installed into my US NTSC C64. I power turn on power. This board does have the 12v jumper set due to my c64 revision.
3. I fire up Quartus. I got 21.x latest as that is what @bodgit I believe said he used when programming this originally. I can swap to 18 if I can find it. I tried 13 and it doesn't have support for the Max 10.
4. I change the Hardware setup to point to the USB-Blaster. Everything appears normal
5. I open the c64mod_2_10.pof file
6. I check Program/Configure
7. I select Start
8. Progress will start but then stalls at 4%. After 20 or 30 seconds, the attempt will timeout and I will be given an error.

One thing that I want to note is that I had to remove the Pot on r27. I originally removed it, added conductors to the legs and re-attached it. The VIC board was hitting the top of it and would not press down all the way with r27 in it's original spot. I then removed it entirely to see if the image stability would change. It got better, but was still off by a lot. I have not reinstalled it while attempting to update the board since I don't even have the JST hooked up from the VIC to the board during programming. I could try that.

What I have tried:
1. Multiple PC's (windows 10 and 11)
2. Different USB Blaster driver installs.
3. Ordered a second "clone" usb-blaster programmer but I get the same results as the original despite being different hardware internally (doing the same thing, but different)
4. I used different USB cables (known good0 and JST connectors.

What I have not tried:
1. Removing the whole board from the C64 and power the VR with a 5v source. Which is how @bodgit described his programming of them in the first place.
2. Remove the jumper to switch from 5v to 12v
3. Reinstalling the VIC as it was and R27.

Pictures of all the things:

[image]

[image]

[image]

[image]

c0pperdragon, 2022-02-11:
The things you did not try 1 - 3 should not make any difference. Especially 2 and 3 are totally unrelated to the FPGA.

But actually it would make sense to check for the stability of the 3.3 volt rails on the FPGA board. If you have an oscilloscope you could check of there is some noise present. Probe both sides of each of L1 - L3. 
I don't know if this could actually influence the flash programming. 

Also try to just program CFM0 (configuration flash memory 0) and not UFM (user flash memory) in one step and see if this alone would succeed. You could then try to programm the other part independently.

When switching to a different quartus version also does not make a difference, I have just one last idea:
As your FPGA board is probably hand-soldered there is on solder contact that could eventually have been missed. It is the ground pad on the underside of the fpga.  
@bodgit , did you also properly solder this one though the hole in the PCB? I don't know what happens if this is unconnected, but it could surely cause problems.

By the way: It is interesting how different the two USB-Blaster devices look at the inside.

bodgit, 2022-02-11:
The board was verified working by myself using the dedicated signal generator as outlined in #47 so it has been flashed successfully in the past however I couldn't remember what version of firmware was programmed.

I have also occasionally had difficulties with programming the boards but often it would start writing and then fail at random percentages of completion rather than just abort at 4%.

For reference, my USB blaster looks different to both of the ones @amerika13 has:

[image]

amerika13, 2022-02-11:
I tried UFM and CFM0 individually and they both stop at 4%. Here are the log messages from the Quartus System window.

`Info (209060): Started Programmer operation at Fri Feb 11 14:19:54 2022
Info (209017): Device 1 contains JTAG ID code 0x031810DD
Warning (16139): Perform full chip erase and program at device index 1
Info (209060): Started Programmer operation at Fri Feb 11 14:19:54 2022
Info (209016): Configuring device index 1
Info (209017): Device 1 contains JTAG ID code 0x031810DD
Info (209007): Configuration succeeded -- 1 device(s) configured
Info (209011): Successfully performed operation(s)
Info (209061): Ended Programmer operation at Fri Feb 11 14:19:56 2022
Info (209024): Programming device 1
Error (209012): Operation failed
Info (209061): Ended Programmer operation at Fri Feb 11 14:20:34 2022
Info (209060): Started Programmer operation at Fri Feb 11 14:21:24 2022
Info (209017): Device 1 contains JTAG ID code 0x031810DD
Warning (16139): Perform full chip erase and program at device index 1
Info (209060): Started Programmer operation at Fri Feb 11 14:21:24 2022
Info (209016): Configuring device index 1
Info (209017): Device 1 contains JTAG ID code 0x031810DD
Info (209007): Configuration succeeded -- 1 device(s) configured
Info (209011): Successfully performed operation(s)
Info (209061): Ended Programmer operation at Fri Feb 11 14:21:26 2022
Info (209024): Programming device 1
Error (209012): Operation failed
Info (209061): Ended Programmer operation at Fri Feb 11 14:22:04 2022
`
@bodgit did tell me that when he programmed the chip, he had all the flags checked. Including Security Bit. I don't know anything about security on these guys, but I hope that didn't lock the chip out from being updated in the future. Is that perhaps the issue?

amerika13, 2022-02-11:
I can choose the Erase option for CFM and UFM and it will say that it was 100% successful. But the Blank-Check option also fails at 4%.

c0pperdragon, 2022-02-11:
This security bit may indeed be something. I actually never used it and I could not even check it anyways, so I never bothered. Same as you, I don't even know what this actually does. It would be a real bummer if this now actually renders your FPGA permanently unusable. 
I am actually at my wits' end here.

amerika13, 2022-02-12:
> @amerika13 Do you have a scope? It would make sense to check how TCK, TMS and TDI look at the board end, and TDO at the programmer end.

I do, but I only got it a couple weeks ago and would need to learn it. Siglent SDS1104X-E 4-Ch 100MHz

Etosha69, 2022-02-14:
Hi al, got the same 4% problem as all the people above. I tried different Quartus Versions on W10, W11 and Linux and also three (!) different USB-Blasters, even the more expensive Terasic one. Problem persists.
The device ist still unusable. Any suggestions?

Etosha69, 2022-02-14:
Additional comment: The security bit on my programming interface of Quartus is not set. By the way erasing the chip works on my side, too, but the blank check also fails at 4%, no matter what USB Blaster or operating system I use.

bodgit, 2022-02-14:
> @bodgit did tell me that when he programmed the chip, he had all the flags checked. Including Security Bit. I don't know anything about security on these guys, but I hope that didn't lock the chip out from being updated in the future. Is that perhaps the issue?

I suspect what is more likely is that I just ticked everything that _could_ be ticked. Given nobody else can enable that security bit, in all likelihood I wouldn't have been able to either.

flegmatik, 2022-02-16:
I also have exact same problem, it fails at 4%. Board was confirmed working before my programming attempt. Tried 2 different USB blasters and two different computers, both with Windows 11 and Quartus version 21.1.

flegmatik, 2022-02-16:
Tried with Quartus 18.1 on Windows 11 and it worked! I'm using Waveshare Altera USB blaster.
https://www.amazon.de/gp/product/B00ID9BAUY/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Etosha69, 2022-02-16:
This are good news!!! :-)
The download of Quartus Lite 18.1 is running. Will post again soon!

amerika13, 2022-02-16:
I haven't gotten to test it yet, but using Quartus 18.1 so far looks to have at least gotten me this far. Programming and verify went through without issue to 100%. I'll be able to test after I get off work.

[image]

bodgit, 2022-02-16:
I would love to know if there's some software other than Quartus that can do the JTAG programming. If you're not actually developing the FPGA code, Quartus is a lot to download just for that purpose.

laubzega, 2022-02-17:
I was able to program the board using OpenOCD 0.11. You will also need:

- The firmware in SVF (which I converted from *.pof using Quartus, but in principle it could be distributed as SVF in the first place).
- If on Windows, the low-level Altera driver needs to be replaced with WinUSB one (easily done using e.g. Zadig - https://zadig.akeo.ie/),

My OpenOCD config file `videomod.cfg`:
<pre>
adapter driver usb_blaster
jtag newtap 10m02 tap -irlen 10 -expected-id 0x031810d
</pre>

My command line
<pre>
openocd.exe -f videomod.cfg -c init -c "svf firmware_2.11_beta1.svf" -c shutdown -l out
</pre>

Initially you may wish to drop `-l out` in order to see the the output with potential issues.

PS. @bodgit please see my private message on BeamRacer forum. ;)

Etosha69, 2022-02-17:
I flashed the device with firmware 2.10 yesterday late evening, testing included. Everything worked well. The component mod is perfectly up and running. Thanks to all for their effort. A success story, so far!
One question remains: Why do you have to use Quartus Version 18.1?!?
I tried Version 13, 15, 16, 17 and 21 (not joking!) - unluckily not the 18. It seems to work only with V18 of Quartus.
This is weird, isn't it?

hugo-lyppens, 2022-02-17:
I would like to report the same findings. Quartus Lite 21.2.0.842 errors out at 4%. Quartus Prime Lite 18.1 successfully programmed 2.10 onto my c0pperdragon composite video mod. Running on Windows 10 x64, using a very inexpensive ($9) USB blaster from Amazon.

Imperious685, 2022-08-12:
I recently built my 2nd one of these, but forgot many of the problems with Quartus 18.1 and my cheap USB-Blaster clone. Essentially Windows just does a BSOD as soon as I plug the Blaster in. I suspect it loads the driver expecting a different IC then crashes as the clone Blaster is not the same hardware. I know there is a way around this by changing the driver but it's a pain when it BSOD's every time.   Then I tried installing Quartus into Linux but it would only install 60% then halted. So I used my Raspberry Pi 3 with OpenOCD.  You can use the config files as above in laubzega's post but make sure the path for the svf file is exact. If You use lower case where there is a capital it will fail with a "no such file or directory error". I had all files in documents but You still need to provide the path for the svf file separately. I've added the svf file here for anyone who doesn't want to install Quartus just to convert the file.
[c64mod210.zip](https://github.com/c0pperdragon/C64-Video-Enhancement/files/9316645/c64mod210.zip)

Etosha69, 2022-08-12:
Hi @Imperious685,
you said you build one! How did you do that? Did you order evrything on your own including the FPGA or did you find a reseller for a kit? Best Regards!

Imperious685, 2022-08-13:
I ordered the boards from PCBWAY and the parts from Digikey then built it myself. You can not currently get the FPGA's from anywhere so unless they become available again You won't be able to get one of these.  This is off topic for this subject anyway so Copperdragon may remove these comments later.

modeler, 2023-09-26:
FPGAs are back in stock at Mouser, I didn't want to believe it myself until I actually had them in my hands. After a wait of two and a half years, I finally finished building this mod, only to hit a horrible issue when attempting to program the FPGA which crapped out at 4%. I was cursing the cheap USB Blaster clone I bought from eBay in 2021, but thanks to this thread it was that later version of Quartus that was the problem.

I am now running Quartus Prime 18.1 on Windows (I'm a Linux user but Debian 12 doesn't play nice with 18.1) and it works perfectly. Once again, thank you all for sharing this info. @c0pperdragon you are a legend!

c0pperdragon, 2023-09-26:
Congratulations! Building the boards successfully is no easy task.

insanity213, 2023-11-07:
I tried using a CH552 based programmer and it couldn't even detect the FPGA.  I've not gotten an FTDI 2232 based programmer that can detect the FPGA but I get an immediate failure when I go to flash.  I've tried a few different DLLs from different FTDI based programmers but the result is always the same.  Not sure what I'm doing wrong.  Has anyone been able to flash this device with one of these FTDI programmers?  Thanks.

Info (209060): Started Programmer operation at Tue Nov 07 21:17:44 2023
Info (209017): Device 1 contains JTAG ID code 0x031810DD
Error (16353): Expected the Max 10 device at index 1 to be of type compact, flash or analog, but detected device of type invalid instead. Ensure that you are using the correct programming file.
Error (209012): Operation failed
Info (209061): Ended Programmer operation at Tue Nov 07 21:17:45 2023

DoctorDan1986, 2023-11-07:
I'm pretty sure I used an Altera Blaster to update mine some years ago.  Could be wrong - it's been a while since I last played with my C64.Sent from my iPhone

insanity213, 2023-11-13:
FWIW, seems that the cheap FTDI and CH552 based USB blasters are just junk.  I grabbed the Terasic blaster from Mouser and it works perfectly.

modeler, 2023-11-13:
The one I have looks like this:

[image]

It was very cheap, about £8 and it works. Your mileage may vary of course,

lomdar67, 2024-05-17:
Tried it with Quartus 18.1 on Windows 11 (latest Version). Unfortunately I can't install the drivers without disabling "driver signature". When I do this and install the driver, Windows 11 crashes the moment I plug the USB Blaster in.

modeler, 2024-05-22:
> Tried it with Quartus 18.1 on Windows 11 (latest Version). Unfortunately I can't install the drivers without disabling "driver signature" in Windows 11. When I do this and install the driver, Windows 11 crashes the moment I plug the USB Blaster in.

I'm running Quartus 18.1 on Windows 11 here with a cheap USB Blaster from eBay and it's working fine:

[image]

[image]

lomdar67, 2024-06-10:
Seems that there are different version of this cheap UBS Blaster and some of them are totally useless unless you flash a custom firmware: https://www.downtowndougbrown.com/2024/06/fixing-a-knockoff-altera-usb-blaster-that-never-worked/