c0pperdragon/C64-Video-Enhancement #81: Analog RF-replacement with S-video for C128

pushead, 2022-03-12:
Hi, 

Hope someone can give me some help here? Not sure if GitHub is the right forum, but I try anyway :-)

I've installed the analog RF-modulator replacement (S-video) with two of my C64's and it works perfect. But I also have a C128 (the slimline version) and a spare RF-modulator replacement module. I read on GVPs webpage that this device could be used with a C128, but I'm not sure if I should solder the "long bord" og "short board" points then? And if the JMPLUM1 should be open or closed? 

Any advice would be much appreciated.

c0pperdragon, 2022-03-12:
The RF modulator of the C128 is the same as on the C64 short boards. So you need to connect the "short board" contacts and leave the JPLUM bridge open.
Also there is no need to connect the GND1 as one of the 8 "short board" pins already provides this.

Bora66, 2022-05-04:
Hi,
This simple analog solution gets quite good reviews, thanks for creating it and making it available for us. I recently salvaged and repaired a C64 Breadbin. It's my first C64 and I am learning a lot on it very quickly. With its original RF modulator, RF output is what it is to be expected, composite out on the DIN socket is a little better, but I can not get a decent picture with s-video (luma-chroma pins of the DIN) no matter what I try (resistors etc) So, I decided to get rid of the RF modulator and replace it with your board. Already ordered the PCB's, and I am about to order the parts. One problem is that I can't find a BOM file here. Most things are self explanatory following the schematic and silk layer, but what is the package size of the SMD resistors and capacitors? And what does JPLUM1 solder jumper do?
Thanks!!

c0pperdragon, 2022-05-04:
The analog-only board uses 0805 packages (imperial) for all passives.
The JPLUM1 is used to attenuate the luma signal from the VIC-II depending on its variant (5V, 12V). The older "long" boards use 12V while the later "short" boards use 5V. Close this jumper for 12V boards.